zephyrproject-rtos/zephyr #9330: network: clean up / implement supervisor to manage net services

mike-scott, 2018-08-07:
The goal of this issue is to discuss and track the implementation of a "network supervisor" for Zephyr.  By implementing a smart network supervisor we can eliminate some of the adhoc code going into various subsystems under subsys/net/lib.  This network supervisor would work the same for applications using net_app or sockets layers.

Steps towards implementation (note: some of these may already be nearly completed):
1. Each of the subsys/net/lib subsections would need to be classified as 1 of the following:
  a. network config layer which is merely a set of startup rules for the network supervisor
  b. network init / management services like DHCP: these influence how the network starts and also can handle changes while running (IE: DHCP re-issues an IP address).
  c. network helper services like DNS which act to assist other network services in some form
  d. network protocol support services like COAP which don't have any threads, but are more a library of code to be used by another entity
  e. network protocols like LwM2M, MQTT and OpenThread which require threads to handle events, etc
2. Each of the above classifications would need implementation (in different forms): DNS is spread around and pretty much a 1 off implementation.  It should work for net_app and sockets layers.  Others like the network protocols with threads would require full APIs w/ callbacks to be managed generically.
3. Once each classification is implemented and each set of libraries adjusted to use it, they can expect to have the network in a state where it can be used when that service is "started".
4. Applications (samples) would also need to be split into 2 different start points:
  a. If needed an application can designate some amount of code to run prior to network initialization.
  b. The normal main() can be started post network similar to how it's done today when subsys/net/lib/conifig/init is used.  This way the sample can also expect a fully functioning network.

mike-scott, 2018-08-07:
@jukkar @pfalcon Adding you here for future discussion

jukkar, 2020-01-28:
Going through old issues. This is a good candidate for closing, currently we have a way to monitor when a network connection is established or closed. Can be re-opened if really needed.

closed 2020-01-28 by jukkar