zephyrproject-rtos/zephyr #9333: Support for STM32 L1-series

postlund, 2018-08-08:
I'm currently looking around for a small RTOS to use in a new project for a customer and Zephyr matches more or less perfectly what I need. My current problem is that the CPU used in the project is based on the STM32L1 series (STM32L152RE to be more exact) and Zephyr does not seem to support that at the moment. So I would like to request support for L1 to be added.

https://www.st.com/en/embedded-software/stm32cubel1.html

postlund, 2018-08-08:
Hi @erwango,

Thanks for the quick reply. I actually started a port last week and it seems quite straight-forward. But I'm new both to the STM32 processor family and zephyr, so it's a slow process (along with my time being limited). But I will try to make some PRs for the things I do and hopefully get things working bit-by-bit. Getting "hello world" working on my development board would be a splendid start.

Thanks again!

Aukstkalnis, 2019-03-13:
As I understand @erwango did not finished STM32L1 MCU support, so I started to work on this.

erwango, 2019-03-14:
@Aukstkalnis, work is actually completed and waiting for 1.15 merge window openning: https://github.com/zephyrproject-rtos/zephyr/pull/12354

Aukstkalnis, 2019-03-14:
Good;-) I spent 1 day to write all GPIO alternate functions for all posible l1 pins and double checking everything😅 So i will check this pull request🙂

erwango, 2019-03-14:
@Aukstkalnis , I don't know what was your ambition, but if you're looking for fun, STM32H7 series are yet to be ported on Zephyr ;-)

Aukstkalnis, 2019-03-14:
My ambition was to add STM32L152C_Discovery board support to zephyr OS and have fun, but there was no STM32L1x support, so I searched open cases and pull request for STM32L1, found this case and didn't check attached cases because their titles did not contain string "STM32L1", so I jumped to conclusion that there is not support for L1 and started working on it :D With this pull request now I only need to add STM32L152C_Discovery board support :)

erwango, 2019-03-14:
So you lost one day, but gained 10 then :-)
This is great, have fun with zephyr on STM32!

Aukstkalnis, 2019-03-14:
I have one question about boards. I am planing to add STM32L152C discovery board support to zephyr. Can I add support on #12354 pull request, test it, then move changes to zephyr master that does not have STM32L1 support and generate pull request? Or should I generate pull request when STM32L1 support will be added?

erwango, 2019-03-14:
You can push a PR with your board support which would include #12354. Then, once #12354 is merged you'll have to rebase and your commits will remain.

closed 2019-04-22 by galak

takumiando, 2020-01-22:
This implementation doesn't work correctly.
https://github.com/zephyrproject-rtos/zephyr/issues/22078

PR #12354 (2019-04-22): Add 96Boards WisTrio board support
This PR adds 96Boards WisTrio board support with STM32L1 SoC series and HAL.

Fixes #9333