zephyrproject-rtos/zephyr #9340: Failed test: kernel.common.errno.thread_context

zephyrbot, 2018-08-08:
***** Booting Zephyr OS 1.12.99 (delayed boot 1000ms) *****
Running test suite context_errno
===================================================================
starting test - test_thread_context
Exception vector: 0x6, cause code: 0x2, parameter 0x2
Address 0x80003800
***** Stack Check Fail! *****
Current thread ID = 0x800078a8
Fatal fault in thread 0x800078a8! Aborting.

nniranjhana, 2018-08-09:
This is related: https://github.com/zephyrproject-rtos/zephyr/issues/9067

nniranjhana, 2018-08-10:
Also, the libraries.encoding tests fail with similar error:

```
***** Booting Zephyr OS 1.12.99 (delayed boot 1000ms) *****
Running test suite lib_json_test
===================================================================
starting test - test_json_encoding
PASS - test_json_encoding
===================================================================
starting test - test_json_decoding
Exception vector: 0x6, cause code: 0x2, parameter 0x2
Address 0x80001800
***** Stack Check Fail! *****
Current thread ID = 0x800058a8
Fatal fault in thread 0x800058a8! Aborting.
```

ruuddw, 2018-08-23:
@nniranjhana 

> "This is related: #9067"

9067 was fixed, I can't test minnowboard but believe the 'errno' tests now pass for ARC.
Can this be closed?

nniranjhana, 2018-08-24:
Okay @ruuddw Currently, I am not able to acquire emsk or minnowboard to test this. I will test on them and close it within a week.

spoorthik, 2018-08-24:
kernel.common.errno.newlib.thread_context test case still fails on minnowboard.

Latest commit on which failure seen: 0d0b221efa6abab94204cd3663641c92a95b1126 

Execution log:
```
CPU Page Fault (error code 0x00000000)
Supervisor thread read address 0x00000000
0x027 Present, Writable, User, Execute Enabled
PTE: 0x00 Non-present, Read-only, Supervisor, Execute Enabled
***** Stack Check Fail! *****
Current thread ID = 0x0010c080
eax: 0x00120240, ebx: 0x0011c0a4, ecx: 0x00102079, edx: 0x00000043
esi: 0x00107d5b, edi: 0x00000000, ebp: 0x0011c004, esp: 0x0011c000
eflags: 0x00210216 cs: 0x0008
eip: 0x001016b6
Fatal fault in essential thread! Spinning...
```

vonhust, 2018-08-24:
@spoorthik  as i know, the latest log is not for ARC, more like for x86.

spoorthik, 2018-08-24:
@vonhust : Right, it is for x86. On ARC, I don't find this issue.

On em_starterkit_em7d_v22,

```
 ***** Booting Zephyr OS v1.13.0-rc1-6-g7983606 (delayed boot 1000ms) *****
Running test suite context_errno
===================================================================
starting test - test_thread_context
PASS - test_thread_context
===================================================================
===================================================================

PROJECT EXECUTION SUCCESSFUL
```

closed 2018-12-07 by nashif