zephyrproject-rtos/zephyr #9352: DFU for nrf52840

IotPanorama, 2018-08-09:
Is it possible to Upgrade a bare metal firmware(not from zephyr, may be from nrf sdk example) through smp server explained in zephyr project ?

cvinayak, 2018-08-09:
@lemrey FYI

cvinayak, 2018-08-09:
Relates to #9055

lemrey, 2018-08-09:
Short answer: no, that is not possible.

The SMP server works with MCUBoot; it won't work with the nRF SDK bootloader.
However, it is possible to migrate from the nRF SDK bootloader to MCUboot and use a built-in firmware update procedure, like the serial recovery mode, to program the SMP server. You would need a signed bootloader image to update bootloader though.

nvlsianpu, 2018-08-09:
I not possible to upgrade bare-metal application using <b>smp</> functionality, unless it is not implemented inside it (and is not). smp server solution's is implemented for zephyr and mynew, not for bare metal (of course you can port it, which dosen't make big sense).

IotPanorama, 2018-08-09:
I have nrf52840 PDK with smp server example running into it. I have also tested DFU by using hello world example of zephyr (having CONFIG_BOOTLOADER_MCUBOOT=y). It also worked as expected. Now i wan't to update the firmware with a simple blinky code (no operating system, no bootloader configuration enabled). will i be able to update that firmware ? at least till the image testing stage.

nvlsianpu, 2018-08-09:
I will be possible once you compile application designed to place from image_0 offset, which vector table shifted by 0x200.

IotPanorama, 2018-08-09:
After changing the address, when i am getting following error after signing the image.
[image]

carlescufi, 2018-08-09:
> Now i wan't to update the firmware with a simple blinky code (no operating system, no bootloader configuration enabled). will i be able to update that firmware ? 

No

IotPanorama, 2018-08-09:
Please explain the reason behind it.

nvlsianpu, 2018-08-09:
@IotPanorama I misses sense of your question before. Once you upgrade running application to the blinky_app you have lost smp_svr functionality as it is not implemented in the blinky - that te reason.

IotPanorama, 2018-08-09:
That's fine . i want to update the image only once. but i am not even being able to sign the blinky image.

nvlsianpu, 2018-08-09:
Look here: https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts#L89
@IotPanorama Your application should be linked into addresses form 0xc200, Your reset vector should start at 0xc200. Your app should take cere about vector relocation if your use any interrupts or expected any exception. Area from 0xc000 to 0xc2000 shouldn't be free.

carlescufi, 2018-10-30:
Closing this since the question was answered

closed 2018-10-30 by carlescufi