zephyrproject-rtos/zephyr #9392: samples/bluetooth/hci_uart ninja flash - UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 360: ordinal not in range(128)

1am, 2018-08-12:
Hi,

I'm trying to compile and flash the [hci_uart](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/bluetooth/hci_uart) example and everything works until the flashing part. It fails with a quite cryptic error and maybe someone will know where I should look for because when inspecting the python west script didn't find anything too obvious. My logs from each step are below:

`$ cmake -GNinja -DBOARD=nrf52_pca10040 ..`

```
CMake Deprecation Warning at /home/user/dev/sdk/zephyr/cmake/app/boilerplate.cmake:38 (cmake_policy):
  The OLD behavior for policy CMP0000 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:7 (include)

-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.4") 
-- Selected BOARD nrf52_pca10040
Zephyr version: 1.12.0
nrf5.conf:1: warning: CONSOLE (defined at drivers/console/Kconfig:10) set more than once. Old value: "y", new value: "n".
nrf5.conf:3: warning: UART_CONSOLE (defined at drivers/console/Kconfig:40) set more than once. Old value: "y", new value: "n".
nrf5.conf:5: warning: SERIAL (defined at drivers/serial/Kconfig:9) set more than once. Old value: "y", new value: "y".
nrf5.conf:10: warning: BT (defined at subsys/bluetooth/Kconfig:10) set more than once. Old value: "y", new value: "y".
Parsing Kconfig tree in /home/user/dev/sdk/zephyr/Kconfig
Using /home/user/dev/sdk/zephyr/boards/arm/nrf52_pca10040/nrf52_pca10040_defconfig as base
Merging nrf5.conf
CMake Warning at /home/user/dev/sdk/zephyr/cmake/toolchain.cmake:14 (message):
  ZEPHYR_GCC_VARIANT is deprecated, please use ZEPHYR_TOOLCHAIN_VARIANT
  instead
Call Stack (most recent call first):
  /home/user/dev/sdk/zephyr/cmake/app/boilerplate.cmake:242 (include)
  CMakeLists.txt:7 (include)

-- Generating zephyr/include/generated/generated_dts_board.h
-- Cache files will be written to: /home/user/dev/sdk/.cache/zephyr
-- The C compiler identification is GNU 6.2.0
-- The CXX compiler identification is GNU 6.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/dev/sdk/zephyr/samples/bluetooth/hci_uart/build
```

`$ ninja`

```
[4/134] Generating always_rebuild
Building for board nrf52_pca10040
[129/134] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       66876 B       512 KB     12.76%
            SRAM:       18364 B        64 KB     28.02%
        IDT_LIST:         148 B         2 KB      7.23%
[134/134] Linking C executable zephyr/zephyr.elf
```

`$ ninja flash`

```
[1/101] Generating always_rebuild
Building for board nrf52_pca10040
[2/2] Flashing nrf52_pca10040
fatal error: ascii b'ing the make.  This is useful for debugging only. With Visual\n// Studio IDE projects all commands are done without /nologo.\nCMAKE_VERBOSE_MAKEFILE:BOOL=FALSE\n\n//If desired, you can build the application usingthe configuration\n// settings specified in an alternate .conf file using this parameter.\n// These settings will override the settings in the application\xe2\x80\x99s\n// .config file or its default .conf file.Multiple files may be\n// listed, e.g. CONF_FILE="prj1.conf prj2.conf"\nCONF_FILE:STRING=nrf5.conf\n\n//Path to a program.\nDTC:FILEPATH=/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc\n\n//If desired, you can build the application using the DT configuration\n// settings specified in an alternate .overlay file using this\n// parameter. These settings will override the settings in the\n// board\'s .dts file. Multiple files may be listed, e.g. DTC_OVERLAY_FILE="dts1.overlay\n// dts2.overlay"\nDTC_OVERLAY_FILE:STRING=/home/user/dev/sdk/zephyr/samples/bluetooth/hci_uart/nrf52_pca10040.overlay\n\n//Single output directory for building all executables.\nEXECUTABLE_OUTPUT_PATH:PATH=\n\n//Git command line client\nGIT_EXECUTABLE:FILEPATH=/usr/bin/git\n\n//Path to a program.\nGPERF:FILEPATH=/usr/bin/gperf\n\n//Single output directory for building all libraries.\nLIBRARY_OUTPUT_PATH:PATH=\n\n//Value Computed by CMake\nNONE_BINARY_DIR:STATIC=/home/user/dev/sdk/zephyr/samples/bluetooth/hci_uart/build\n\n//Value Computed by CMake\nNONE_SOURCE_DIR:STATIC=/home/user/dev/sdk/zephyr/samples/bluetooth/hci_uart\n\n//Path to a program.\nOPENOCD:FILEPATH=/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd\n\n//Path to a program.\nPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3\n\n//Default runner for debugging\nZEPHYR_BOARD_DEBUG_RUNNER:STRING=jlink\n\n//Default runner for flashing binaries\nZEPHYR_BOARD_FLASH_RUNNER:STRING=nrfjprog\n\n//Runner-specific arguments for jlink\nZEPHYR_RUNNER_ARGS_jlink:STRING=--dt-flash=y;--device=nrf52;--speed=4000\n\n//Runner-specific arguments for nrfjprog\nZEPHYR_RUNNER_ARGS_nrfjprog:STRING=--nrf-family=NRF52\n\n//Board definition directory\nZEPHYR_RUNNER_CONFIG_BOARD_DIR:STRING=/home/user/dev/sdk/zephyr/boards/arm/nrf52_pca10040\n\n//Path to GDB binary, if applicable\nZEPHYR_RUNNER_CONFIG_GDB:STRING=/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gdb\n\n//Path to kernel image as raw binary\nZEPHYR_RUNNER_CONFIG_KERNEL_BIN:STRING=/home/user/dev/sdk/zephyr/samples/bluetooth/hci_uart/build/zephyr/zephyr.bin\n\n//Path to kernel image in ELF format\nZEPHYR_RUNNER_CONFIG_KERNEL_ELF:STRING=/home/user/dev/sdk/zephyr/samples/bluetooth/hci_uart/build/zephyr/zephyr.elf\n\n//Path to kernel image in Intel Hex format\nZEPHYR_RUNNER_CONFIG_KERNEL_HEX:STRING=/home/user/dev/sdk/zephyr/samples/bluetooth/hci_uart/build/zephyr/zephyr.hex\n\n//Path to openocd binary, if applicable\nZEPHYR_RUNNER_CONFIG_OPENOCD:STRING=/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd\n\n//Path to add to openocd search path, if applicable\nZEPHYR_RUNNER_CONFIG_OPENOCD_SEARCH:STRING=/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts\n\n//Zephyr SDK install directory\nZEPHYR_SDK_INSTALL_DIR:PATH=/opt/zephyr-sdk\n\n//Zephyr toolchain variant\nZEPHYR_TOOLCHAIN_VARIANT:STRING=zephyr\n\n//Value Computed by CMake\nZephyr-Kernel_BINARY_DIR:STATIC=/home/user/dev/sdk/zephyr/samples/bluetooth/hci_uart/build/zephyr\n\n//Value Computed by CMake\nZephyr-Kernel_SOURCE_DIR:STATIC=/home/user/dev/sdk/zephyr\n\n//Dependencies for the target\napp_LIB_DEPENDS:STATIC=general;zephyr_interface;\n\n//Dependencies for target\narch__arm__core_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\narch__arm__core__cortex_m_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\narch__arm__core__cortex_m__mpu_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\ndrivers__entropy_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\ndrivers__gpio_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\ndrivers__serial_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\nkernel_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\noffsets_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\nsubsys__bluetooth__common_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\nsubsys__bluetooth__controller_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\nsubsys__bluetooth__host_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\nsubsys__net_LIB_DEPENDS:STATIC=\n\n//Dependencies for target\nzephyr_LIB_DEPENDS:STATIC=\n\n\n########################\n# INTERNAL cache entries\n########################\n\n//ADVANCED property for variable: CMAKE_AR\nCMAKE_AR-ADVANCED:INTERNAL=1\n//ADVANCED property for variable: CMAKE_ASM_COMPILER\nCMAKE_ASM_COMPILER-ADVANCED:INTERNAL=1\nCMAKE_ASM_COMPILER_WORKS:INTERNAL=1\n//ADVANCED property for variable: CMAKE_ASM_FLAGS\nCMAKE_ASM_FLAGS-ADVANCED:INTERNAL=1\n//ADVANCED property for variable: CMAKE_ASM_FLAGS_DEBUG\nCMAKE_ASM_FLAGS_DEBUG-ADVANCED:INTERNAL=1\n//ADVANCED property for variable: CMAKE_ASM_FLAGS_MINSIZEREL\nCMAKE_ASM_FLAGS_MINSIZEREL-ADVANCED:INTERNAL=1\n//ADVANCED property for variable: CMAKE_ASM_FLAGS_RELEASE\nCMAKE_ASM_FLAGS_RELEASE-ADVANCED:INTERNAL=1\n//ADVANCED property for variable: CMAKE_ASM_FLAGS_RELWITHDEBINFO\nCMAKE_ASM [... cut ...]
My nrfutil, nrfjprog and so on are in the path as I'm working with them every day and i can interact with the flashed board through them, reset, flash .hex normally. For some reason only `ninja flash` fails and does this very early so i don't expect that it even gets to the part where there is any flashing or other interaction with hardware part. 
The CMakeCache.txt file looks pretty ok and the `ZEPHYR_RUNNERS` is set to `ZEPHYR_RUNNERS:INTERNAL=nrfjprog;jlink` so pretty much what one could expect

1am, 2018-08-12:
Same thing for python 3.6

vanwinkeljan, 2018-08-12:
Are your locals set correctly?
Could you try again after setting LC_ALL to UTF-8 (e.g. export LC_ALL=en_US.UTF-8)

1am, 2018-08-12:
You are right. Thank you for your help.

closed 2018-08-12 by 1am