zephyrproject-rtos/zephyr #9402: samples/drivers/watchdog fails on freedom_k64f

spoorthik, 2018-08-13:
Execution log:

```
Start watchdog test
***** BUS FAULT *****
  Instruction bus error
  NXP MPU error, port 4
    Mode: Supervisor, Instruction Address: 0x9f38018
    Type: Read, Master: 0, Regions: 0xc000
***** Hardware exception *****
Current thread ID = 0x200000d4
Faulting instruction address = 0x9f38018
Fatal fault in essential thread! Spinning...
```

Board: freedom_k64f
Architecture: ARM

Latest commit: d71a1dfefd3ea0b7cefa3a3d1958a9238b5b1861

MaureenHelm, 2018-08-13:
This sample needs to be updated to the new watchdog API. See #8215 

@kl-cruz Would it be possible to update the sample for 1.13?

kl-cruz, 2018-08-14:
I'll try to update it today, but I do not want to promise anything. There is "Feature Merge Window Close" tomorrow

nashif, 2018-08-20:
this is now fixed, however quark is still broken, but that could be the driver.

closed 2018-08-20 by nashif