zephyrproject-rtos/zephyr #9432: Overriding `LOG_LEVEL` could crash the firmware

Olivier-ProGlove, 2018-08-14:
Here is a test based on `sample/hello_world` that crashes:

```
#include <zephyr.h>

#define LOG_MODULE_NAME main
#define LOG_LEVEL 0
#include <logging/log.h>
LOG_MODULE_REGISTER();

void main(void)
{
	LOG_DBG("Hello World!");
}
```

with `prj.conf`:

```
# Enable console
CONFIG_CONSOLE=y

# Enabling logging
CONFIG_LOG=y
CONFIG_LOG_RUNTIME_FILTERING=y

# Override lowest log level to DEBUG(=4)
CONFIG_LOG_OVERRIDE_LEVEL=4
```

- Commenting `CONFIG_LOG_OVERRIDE_LEVEL=4` does not make the firmware crash
- Commenting `CONFIG_LOG_RUNTIME_FILTERING=y` does not make the firmware crash
- Commenting `#define LOG_LEVEL 0` does not make the firmware crash

dleach02, 2019-02-21:
I tried to reproduce this but was not able to. Can @Olivier-ProGlove, retest and put the exact code/prj config values that crashes?

nashif, 2019-03-03:
not able to reproduce

closed 2019-03-03 by nashif