zephyrproject-rtos/zephyr #9452: Error parsing DTS 'compatible' property list

nategraff-sifive, 2018-08-15:
In `scripts/dts/extract/globals.py`, the function `get_compat()` returns a list instead of a single string if a node doesn't have the `compatible` property and its parent's `compatible` property is a list. This causes `generate_node_definitions()` to error.

## Current implementation of `get_compat()`
```
def get_compat(node_address):
    compat = None

    try:
        if 'props' in reduced[node_address].keys():
            compat = reduced[node_address]['props'].get('compatible')

        if isinstance(compat, list):
            compat = compat[0]

        if compat == None:
            compat = find_parent_prop(node_address, 'compatible')

    except:
        pass

    return compat
```

## Suggested fix:
Swap the order of the second and third conditionals so that if the parent's `compatible` property is a list, `get_compat()` returns the first element.
```
def get_compat(node_address):
    compat = None

    try:
        if 'props' in reduced[node_address].keys():
            compat = reduced[node_address]['props'].get('compatible')

        if compat == None:
            compat = find_parent_prop(node_address, 'compatible')

        if isinstance(compat, list):
            compat = compat[0]

    except:
        pass

    return compat
```

galak, 2018-08-15:
Got an example you are hitting?  No issue with the change, just helpful to understand how you are hitting the issue.

nategraff-sifive, 2018-08-15:
Here's a snippet of a DTS file that would cause the issue.

The `global-external-interrupts` node doesn't have a `compatible` property, and its parent `soc` node has a list for its `compatible` property.

```
...
    soc {                                                                                                                                                                                                                                                  
        #address-cells = <1>;                                                                                                                                                                                                                                   
        #size-cells = <1>;                                                                                                                                                                                                                                      
        compatible = "SiFive,FE310G-soc", "fe310-soc", "sifive-soc", "simple-bus";                                                                                                                                                                              
        ranges;                                                                                                                                                                                                                                                 
        clint: clint@2000000 {                                                                                                                                                                                                                                     
            compatible = "riscv,clint0";                                                                                                                                                                                                                        
            interrupts-extended = <&hlic 3 &hlic 7>;                                                                                                                                                                                                                
            reg = <0x2000000 0x10000>;                                                                                                                                                                                                                          
            reg-names = "control";                                                                                                                                                                                                                              
        };
        global-external-interrupts {                                                                                                                                                                                                                        
            interrupt-parent = <&plic>;                                                                                                                                                                                                                           
            interrupts = <1 2 3 4>;                                                                                                                                                                                                                             
        };
...
```

Admittedly, we might need to modify the DTS so that we don't hit this edge case anyway, but it's clear that this edge case breaks the expected return type of `get_compat()`.

closed 2018-08-16 by nategraff-sifive