zephyrproject-rtos/zephyr #9459: tests/posix/timer fails on nRF51/52

nniranjhana, 2018-08-16:
The test in tests/posix/timer results in _assertion failure_.

**Arch:** arm
**Board:** nrf51/52 - nrf51_pca10028:arm, nrf52_pca10040:arm, nrf52840_pca10056:arm

**Error console log:**
```
 ***** Booting Zephyr OS 1.12.99 (delayed boot 1000ms) *****
 Running test suite test_posix_timer
 ===================================================================
 starting test - test_timer
 POSIX timer test
 Timer fires every 0 secs and  100000000 nsecs
 Time remaining to fire 0 secs and  888000000 nsecs
 Handler Signal value :20 for 1 times
 Handler Signal value :20 for 2 times
 Handler Signal value :20 for 3 times
 Handler Signal value :20 for 4 times
 Handler Signal value :20 for 5 times
 Handler Signal value :20 for 6 times
 Handler Signal value :20 for 7 times
 Handler Signal value :20 for 8 times
 Handler Signal value :20 for 9 times
 Handler Signal value :20 for 10 times
 Handler Signal value :20 for 11 times
 
 Assertion failed at .. /tests/posix/timer/src/main.c:87: test_timer: (total_secs_timer not equal to secs_elapsed)
 POSIX timer test has failed
 FAIL - test_time
```

ManojSubbarao, 2018-08-30:
@rgundi Please take a look at this issue.

nniranjhana, 2018-08-30:
Updated with the right commit after which test started failing.

rgundi, 2018-08-30:
Looks like the test case needs to be modified. Looking into it.

PR #9745 (2018-08-31): kernel: Improve precision of ticks and ms conversions
The following 2 improvements are contained in this patch:

- When converting from ms to ticks, instead of using hardware cycles
  per tick, use hardware cycles per second. This ensures that the
  multiplication is done before the division, increasing precision.
- When converting from ticks to ms, instead of using cycles per tick
  and cycles per sec, use ticks per sec. This too increases the
  precision.

The concept is to make the dividend as large as possible compared to the
divisor in order to lose as little precision as possible.

Fixes [... cut]
commit c7d27344 (2018-08-31): kernel: Improve precision of ticks and ms conversions

The following 2 improvements are contained in this patch:

- When converting from ms to ticks, instead of using hardware cycles
  per tick, use hardware cycles per second. This ensures that the
  multiplication is done before the division, increasing precision.
- When converting from ticks to ms, instead of using cycles per tick
  and cycles per sec, use ticks per sec. This too increases the
  precision.

The concept is to make the dividend as large as possible compared to the
divisor in order to lose as little precision as possible.

Fixe [... cut]
closed 2018-08-31 by nashif