zephyrproject-rtos/zephyr #9463: bt_le_scan_start Fails with Error -5 after 128 scan start/stop cycles

DeclanTraill, 2018-08-16:
I have a basic project that scans for 5 seconds, then stops for one second - then repeats forever.
After exactly 128 cycles, the 129th scan start (and all after that) fails with error code -5 (I think this is POSIX:       #define EIO 5		/* I/O error */).

Given the number of time it works (128) it seems like a resource or memory may have run out?

Here is the debug log (attached) showing the problem:   Any ideas what might be wrong & how to fix it? 

Thanks...

[Zephyr_Scan_Start_Stop_Error.txt](https://github.com/zephyrproject-rtos/zephyr/files/2292440/Zephyr_Scan_Start_Stop_Error.txt)

Regards,
Declan

nashif, 2018-08-21:
@jhedberg  fyi

cvinayak, 2018-08-22:
There seems to be a bug somewhere, I need time til 24/8 to reproduce this, before i can say anything.

jhedberg, 2018-08-22:
Copying info here from the mailing list:

The command that's failing is HCI_Set_Random_Address, and it's failing with status 0x0c, i.e. "Command Disallowed".

jhedberg, 2018-08-30:
@DeclanTraill are you also doing simultaneous advertising like @cvinayak suspects? Are you doing passive or active scanning? It's only active scanning that requires touching the local random address, so if you don't need to do that then switching to passive scanning may work around your issue. Let's try to keep all the discussion here rather than the duplicate thread on the mailing list.

jhedberg, 2018-08-30:
@cvinayak regarding the conflict with advertising, what I'd do is not change the random address if advertising is active using it. I assume this issue only concerns random address based advertising, i.e. advertising with public address would not prevent changing the random one? Anyway, I think this policy should be ok, since the reason why active scanning tries to change the address is in order not to expose the local identity (so either NRPA or RPA). What ever address we're advertising with is already being exposed, so it should be fine to just use that for the scan requests.

cvinayak, 2018-08-30:
@jhedberg I have a bad feeling, I have been review the code path, and I am not able to figure out why the scan stop/start fails on set_random_address only after the RPA timeout has expired... from the logs, it appears, the advertising is stopped and restarted, but set_random_address is failing at this point itself.

cvinayak, 2018-08-30:
@DeclanTraill FYI #9728

DeclanTraill, 2018-08-31:
Great work – thanks guys…
 
Regards,
Declan

 

Email from setec.com.au does not necessarily represent the official
policy of SETEC Pty Ltd.
See http://www.setec.com.au/emaildisclaimer20060629.html for details.

nashif, 2019-02-19:
any update on this? @cvinayak @jhedberg ?

carlescufi, 2019-03-26:
@joerchan @DeclanTraill is this still an issue, do we need https://github.com/zephyrproject-rtos/zephyr/pull/9728 to get in or is this already fixed elsewhere?

cvinayak, 2019-03-26:
@carlescufi I pushed an update based on review comments. The PR still needs verification by someone else, to reproduce without the PR and check for resolution of the issue with the PR applied.

PR #9728 (2019-03-28): Bluetooth: Fix scan failed to start due to RPA timeout
Updating the Resolvable Private Address when advertising and
active scanning in progress fails and clears the RPA_VALID
flag; making the next bt_le_scan_start while continueing to
advertise to fail.

This is fixed by keeping the RPA_VALID flag remain set.
Stopping and starting active scanning to update RPA can be
implemented in a separate commit.

Fixes #9463.
commit 3353c9f3 (2019-03-28): Bluetooth: Fix scan failed to start due to RPA timeout

Updating the Resolvable Private Address when advertising and
active scanning in progress fails and clears the RPA_VALID
flag; making the next bt_le_scan_start while continuing to
advertise to fail.

This is fixed by keeping the RPA_VALID flag remain set.
Stopping and starting active scanning to update RPA can be
implemented in a separate commit.

Fixes #9463.
closed 2019-03-28 by carlescufi