zephyrproject-rtos/zephyr #9468: tests/kernel/mem_pool/mem_pool_concept/testcase.yaml#kernel.memory_pool fails on nrf52840_pca10056, nrf52_pca10040 and  nrf51_pca10028

spoorthik, 2018-08-16:
Execution log:
```
Running test suite mpool_concept
===================================================================
starting test - test_mpool_alloc_wait_prio

    Assertion failed at /home/kspoorth/work/latest_zephyr/tests/kernel/mem_pool/mem_pool_concept/src/test_mpool_alloc_wait.c:24: tmpool_alloc_wait_timeout: k_mem_pool_alloc(&mpool1, &block, BLK_SIZE_MIN, TIMEOUT) == -EAGAIN is false

FAIL - test_mpool_alloc_wait_prio
===================================================================
starting test - test_mpool_alloc_size_roundup

    Assertion failed at /home/kspoorth/work/latest_zephyr/tests/kernel/mem_pool/mem_pool_concept/src/test_mpool_alloc_size.c:36: test_mpool_alloc_size_roundup: k_mem_pool_alloc(&mpool1, &block[i], TEST_SIZE, K_NO_WAIT) == 0 is false

FAIL - test_mpool_alloc_size_roundup
===================================================================
starting test - test_mpool_alloc_merge_failed_diff_size
PASS - test_mpool_alloc_merge_failed_diff_size
===================================================================
starting test - test_mpool_alloc_merge_failed_diff_parent

    Assertion failed at /home/kspoorth/work/latest_zephyr/tests/kernel/mem_pool/mem_pool_concept/src/test_mpool_merge_fail_diff_parent.c:33: test_mpool_alloc_merge_failed_diff_parent: k_mem_pool_alloc(&mpool1, &block[i], BLK_SIZE_MIN, K_NO_WAIT) == 0 is false

FAIL - test_mpool_alloc_merge_failed_diff_parent
===================================================================
===================================================================
PROJECT EXECUTION FAILED
```

Steps to reproduce:
cd tests/kernel/mem_pool/mem_pool_concept
mkdir build && cd build
cmake -DBOARD=nrf51_pca10028 .. && make flash

Latest commit: afa0e0026fb94d37f50dc53f9d0981b1dc5c306b

Platforms tested: nrf52840_pca10056, nrf52_pca10040 and  nrf51_pca10028

nashif, 2018-08-30:
@carlescufi fyi

carlescufi, 2018-08-30:
@ioannisg fyi, let's take a look at this together

PR #9745 (2018-08-31): kernel: Improve precision of ticks and ms conversions
The following 2 improvements are contained in this patch:

- When converting from ms to ticks, instead of using hardware cycles
  per tick, use hardware cycles per second. This ensures that the
  multiplication is done before the division, increasing precision.
- When converting from ticks to ms, instead of using cycles per tick
  and cycles per sec, use ticks per sec. This too increases the
  precision.

The concept is to make the dividend as large as possible compared to the
divisor in order to lose as little precision as possible.

Fixes [... cut]
commit c7d27344 (2018-08-31): kernel: Improve precision of ticks and ms conversions

The following 2 improvements are contained in this patch:

- When converting from ms to ticks, instead of using hardware cycles
  per tick, use hardware cycles per second. This ensures that the
  multiplication is done before the division, increasing precision.
- When converting from ticks to ms, instead of using cycles per tick
  and cycles per sec, use ticks per sec. This too increases the
  precision.

The concept is to make the dividend as large as possible compared to the
divisor in order to lose as little precision as possible.

Fixe [... cut]
closed 2018-08-31 by nashif