zephyrproject-rtos/zephyr #9470: LWM2M: TLV encoding of read result is wrong

therealprof, 2018-08-16:
When doing a `Read` on a an object without object instance identifier, the reply is supposed to use an object instance TLV as stated in the OMA TS: 

```
For simplicity and to prevent any ambiguity on Object Instance Identity, when the Object Instance ID is not specified in the request, the Object Instance TLV MUST be used, whatever the number of Object Instances (1 or more) to return to the LwM2M Server.
```

Zephyr currently does not do this. If I e.g. do a `Read` on `/3` the result TLV directly starts with the first resource identifier instead of the Object Instance TLV for the object instance ID 0.

mike-scott, 2018-08-16:
@therealprof Ah interesting.  I'll take a look later.

Do we want this in v1.13?

therealprof, 2018-08-16:
> Do we want this in v1.13?

Don't know. Depends on how serious people are taking LwM2M. ;) I've implemented a workaround today in our server software because another (commercial) product is having the exact same problem but it would be great to be more compliant, at least with the old spec.

mike-scott, 2018-08-22:
@therealprof Ah, we get this behavior because we never check that the path->level is > 1 (which would indicate an object instance id was passed in)

The intended response for a request on /3 was supposed to be ENOENT atm.   Instead we default to 0 which acts like we requests /3/0

In any case, none of the above responses are actually correct. :)  We need to handle the larger object instance use-case correctly.

closed 2018-08-30 by nashif