zephyrproject-rtos/zephyr #9486: sanitycheck filter rules does not work

hakehuang, 2018-08-17:
when I run sanitycheck script 
./scripts/sanitycheck --dry-run -p lpcxpresso54114_m4
./scripts/sanitycheck --dry-run -p frdm_kw41z
I find many unsupported samples/tests are mark as enabled on this platform.
the questioned samples and test applications are but may not limited to those

- Kernal_fatal
- mem_protect_protection
- mpu_stack_guard_test
- arm_irq_vector_table
- gen_isr_table
- mem_domain_apis_test
- tests/kernel/mem_protect/userspace
- tests/kernel/mem_protect/obj_validation
- samples/boards/quark_se_c1000/power_mgr
- tests/power/power_states
- tests/kernel/tickless/tickless

the root cause seems that the filter in each testcase.yaml/sample.yaml does not work. 
below two options seems can not used to filter out applications
platform_whitelist:
platform_exclude:

hakehuang, 2018-08-17:
cc @MaureenHelm, please help to label this issue. Thanks

nashif, 2018-08-17:
@hakehuang where do you see that "tests/power/power_states" is enabled when you run the commands above?

Try:
`./scripts/sanitycheck -p frdm_kw41z --save-tests frdm_kw41z.tests`

look at the generated file, it should not have the above tests.

nashif, 2018-08-31:
@hakehuang just because it is in the makefile, does not mean it is going to run. This is done to verify Kconfig filters and compare with what the test is expecting, it will not run the test on those platforms though.

./scripts/sanitycheck -p frdm_kw41z --save-tests frdm_kw41z.tests gives you the final list after filtering.

nashif, 2018-09-05:
> Thanks @nashif, but what is the meaning of compiling an application which can not run? why not remove then totally?

we do not compile, we just create the kconfig for the board and check if some kconfig options are available from the board, if no, we exclude it, if yes and if the test has a filter on kconfig, it is added.

hakehuang, 2018-09-06:
I see. Thanks