zephyrproject-rtos/zephyr #9494: Nordic nrf52810_pca10040 is missing default bluetooth configuration options

thedjnK, 2018-08-17:
When building sample applications with Bluetooth using the nrf52810_pca10040 board configuration, an error is emitted:

    C:/zephyr/drivers/bluetooth/hci/h4.c: In function '_bt_uart_init':
    C:/zephyr/drivers/bluetooth/hci/h4.c:462:30: error: 'CONFIG_BT_UART_ON_DEV_NAME' undeclared (first use in this function);
    did you mean 'CONFIG_UART_0_NAME'?
      h4_dev = device_get_binding(CONFIG_BT_UART_ON_DEV_NAME);
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                  CONFIG_UART_0_NAME
    C:/zephyr/drivers/bluetooth/hci/h4.c:462:30: note: each undeclared identifier is reported only once for each function it appears in
    [34/128] Building C object zephyr/CMak....dir/drivers/timer/nrf_rtc_timer.c.obj
    ninja: build stopped: subcommand failed.

Selecting the option Bluetooth -> Bluetooth Controller fixes this issue but it should be added to the default configuration as it is with the nrf52840_pca10056 board.

ioannisg, 2018-08-17:
>Selecting the option Bluetooth -> Bluetooth Controller fixes this issue but it should be added to the default configuration as it is with the nrf52840_pca10056 board.

 AFAICS, it is documented _why_ the BT options are not enabled by default n 52810.

# Bluetooth not enabled by default on nRF52810 due to RAM limitations when
# running the default set of kernel tests.
# Enable this on your prj.conf to include Bluetooth support

SebastianBoe, 2018-08-17:
I get that we don't enable BT on 52810. But IMHO when BT is enabled by the app we should default to using BT_CTLR and not BT_UART as the BT samples are doing.

PR: #9499

closed 2018-08-17 by nashif

SebastianBoe, 2018-08-20:
@thedjnK : Thank you for reporting.

PR #9499 (2018-08-17): kconfig: BT: Default to using BT_CTLR when BT
As reported in issue #9494, BT samples are using the off-chip
UART-based BT controller on 52810. But the on-chip BT controller is
supported on 52810, so this is the reasonable default when BT is
enabled.

This patch enables BT_CTLR by default when BT is enabled.

This resolves #9494.
commit 9e18b4f0 (2018-08-17): kconfig: BT: Default to using BT_CTLR when BT

As reported in issue #9494, BT samples are using the off-chip
UART-based BT controller on 52810. But the on-chip BT controller is
supported on 52810, so this is the reasonable default when BT is
enabled.

This patch enables BT_CTLR by default when BT is enabled.

This resolves #9494.