zephyrproject-rtos/zephyr #9498: Invalid argument saved on ISR_CONNECT

MarekPieta, 2018-08-17:
ISR_CONNECT macro is used to register interrupt service routine. One of its arguments is a parameter that should be passed to isr callback when interrupt is called. Unfortunately this value can be wrongly recorded in some cases.
When address to object in RAM (from bss section) is passed as a parameter to ISR_CONNECT and there is a custom section in RAM present before the bss, address of this object is wrong.
It seems that address to object is recorded as an offset from beginning of bss section, or that it is recorded before linker takes into account whole memory layout (including custom section before bss that was mentioned above).
The problem can lead to system data corruption and in consequence to kernel panic.
Sometimes data corruption can be not evident and kernel can run without any apparent issues (problem remains hidden). This depends on how the parameter (object recorded address) is altered from its real value. 

When problem was replicated for the first time, corrupted data was seen in Nordic’s RNG driver that passes address to driver’s data to ISR_CONNECT.

**Replication**
Issue found on ARM - nrf52_pca10040 board.

Problem can be reproduced with bluetooth/beacon/ on nrf52_pca10040 by adding following lines to main.c  (error was observed using RTT console which was also enabled):
`char volatile error_variable[32] __used __attribute__((__section__(".i_make_error"))) = {0};`

To make sure the custom section is not opt-out add reference to it form any function (e.g. `main()`):
`error_variable[2] = 3;`

Received error:
```
***** Booting Zephyr OS v1.12.0-1476-gba6763a *****
Sta***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0xc7fc
***** Hardware exception *****
Current thread ID = 0x200013ec
Faulting instruction address = 0xb272
Fatal fault in ISR! Spinning...
```

nashif, 2018-08-17:
you mean IRQ_CONNECT?

MarekPieta, 2018-08-20:
Yes, I've changed it already.

dcpleung, 2018-09-12:
The problem is how the linker places the orphan section in the ELF file.

With the prebuilt ELF, the ".i_make_error" is placed after ".intList" and not after data or bss.
```
$ readelf -S zephyr/zephyr_prebuilt.elf 
There are 24 section headers, starting at offset 0x31b44:

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] text              PROGBITS        00000000 0000b4 001522 00 WAX  0   0  4
  [ 2] devconfig         PROGBITS        00001524 0015d8 000054 00  WA  0   0  4
  [ 3] rodata            PROGBITS        00001578 00162c 00048c 00   A  0   0  4
  [ 4] app_smem          PROGBITS        20000000 001b54 000000 00   W  0   0  1
  [ 5] bss               NOBITS          20000000 001b30 000218 00  WA  0   0  8
  [ 6] noinit            NOBITS          20000218 001b30 000d00 00  WA  0   0  8
  [ 7] datas             PROGBITS        20000f18 001ab8 000020 00  WA  0   0  4
  [ 8] initlevel         PROGBITS        20000f38 001ad8 000054 00  WA  0   0  4
  [ 9] .intList          PROGBITS        20010000 001b2c 000008 00  WA  0   0  4
  [10] .i_make_error     PROGBITS        20010008 001b34 000020 00  WA  0   0  1
  [11] .debug_info       PROGBITS        00000000 001b54 015ed4 00      0   0  1
  [12] .debug_abbrev     PROGBITS        00000000 017a28 004e6b 00      0   0  1
  [13] .debug_aranges    PROGBITS        00000000 01c898 000868 00      0   0  8
  [14] .debug_ranges     PROGBITS        00000000 01d100 000d60 00      0   0  8
  [15] .debug_line       PROGBITS        00000000 01de60 006dc0 00      0   0  1
  [16] .debug_str        PROGBITS        00000000 024c20 002ced 01  MS  0   0  1
  [17] .comment          PROGBITS        00000000 02790d 000011 01  MS  0   0  1
  [18] .ARM.attributes   ARM_ATTRIBUTES  00000000 02791e 000031 00      0   0  1
  [19] .debug_frame      PROGBITS        00000000 027950 000f64 00      0   0  4
  [20] .debug_loc        PROGBITS        00000000 0288b4 0049ef 00      0   0  1
  [21] .shstrtab         STRTAB          00000000 031a59 0000eb 00      0   0  1
  [22] .symtab           SYMTAB          00000000 02d2a4 002820 10     23 323  4
  [23] .strtab           STRTAB          00000000 02fac4 001f95 00      0   0  1
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required), G (group), T (TLS),
  C (compressed), x (unknown), o (OS specific), E (exclude),
  y (purecode), p (processor specific)
```

And with the finally ELF, it is placed before bss:
```
$ readelf -S zephyr/zephyr.elf 
There are 23 section headers, starting at offset 0x31de4:

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] text              PROGBITS        00000000 0000d4 001522 00 WAX  0   0  4
  [ 2] .i_make_error     PROGBITS        20000000 0015f6 000020 00  WA  0   0  1
  [ 3] devconfig         PROGBITS        00001544 001618 000054 00  WA  0   0  4
  [ 4] rodata            PROGBITS        00001598 00166c 00048c 00   A  0   0  4
  [ 5] app_smem          PROGBITS        20000020 001b6c 000000 00   W  0   0  1
  [ 6] bss               NOBITS          20000020 001b70 000218 00  WA  0   0  8
  [ 7] noinit            NOBITS          20000238 001b70 000d00 00  WA  0   0  8
  [ 8] datas             PROGBITS        20000f38 001af8 000020 00  WA  0   0  4
  [ 9] initlevel         PROGBITS        20000f58 001b18 000054 00  WA  0   0  4
  [10] .debug_info       PROGBITS        00000000 001b6c 015fb4 00      0   0  1
  [11] .debug_abbrev     PROGBITS        00000000 017b20 004f20 00      0   0  1
  [12] .debug_aranges    PROGBITS        00000000 01ca40 000880 00      0   0  8
  [13] .debug_line       PROGBITS        00000000 01d2c0 006e8a 00      0   0  1
  [14] .debug_str        PROGBITS        00000000 02414a 002d33 01  MS  0   0  1
  [15] .comment          PROGBITS        00000000 026e7d 000011 01  MS  0   0  1
  [16] .ARM.attributes   ARM_ATTRIBUTES  00000000 026e8e 000031 00      0   0  1
  [17] .debug_ranges     PROGBITS        00000000 026ec0 000d60 00      0   0  8
  [18] .debug_frame      PROGBITS        00000000 027c20 000f64 00      0   0  4
  [19] .debug_loc        PROGBITS        00000000 028b84 0049ef 00      0   0  1
  [20] .shstrtab         STRTAB          00000000 031d00 0000e2 00      0   0  1
  [21] .symtab           SYMTAB          00000000 02d574 002800 10     22 322  4
  [22] .strtab           STRTAB          00000000 02fd74 001f8c 00      0   0  1
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required), G (group), T (TLS),
  C (compressed), x (unknown), o (OS specific), E (exclude),
  y (purecode), p (processor specific)
```

Since the isr_tables.c is created using zephyr_prebuilt.elf, and the device pointers as ISR arguments are using the addresses in the prebuilt. However, when the final ELF is generated, the bss/data sections gets pushed away, which moves all the variables to another location. The device pointers as ISR arguments are thus no longer valid.

This is not something that can be fixed in Zephyr in a generic way, as this behavior is highly dependent on compiler/linker behavior. The only thing the developer can do is to enable "CONFIG_CUSTOM_RWDATA_LD" and provides additional linker instructions to properly place the custom sections.

carlescufi, 2018-09-13:
@MarekPieta have you hit this in a real example?
CC @pizi-nordic @ioannisg @SebastianBoe

SebastianBoe, 2018-09-14:
This is arguably user-error. It is a bit weird to have a character array that is not placed in the text section.

In this particular case the user should have done

__attribute__((__section__(".text.i_make_error")))

and everything would have been fine.

If, for whatever reason, the user wants to place his custom section outside of the text area he had better tell the linker where to put it, as @dcpleung shows.

So in the near future, I do not see this creating any problems. But we should try to find better error handling. How can the user discover that orphaned sections are not supported?

I wonder if we should turn orphans into errors. See https://github.com/SebastianBoe/zephyr/commit/4351482ecd6388c5f2791b8a12d3546f23ae8ffe for how this could be done.

In that patch, we get the error:

`/home/sebo/zephyr-sdk-0.9.3/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: error: unplaced orphan section `.i_make_error' from `libapp.a(main.c.obj)'.`

which is much easier for a user to deal with than a runtime error.

pizi-nordic, 2018-09-14:
> This is arguably user-error. It is a bit weird to have a character array that is not placed in the text section.

No, it is not. For example the new logger subsystem uses dedicated sections in RAM and Flash to maintain global list of modules (in flash) and its settings (in ram, as this part might be modified in runtime).

SebastianBoe, 2018-09-14:
That is no problem, just place this information at

.text.logger.xyz
.text.logger.abc

.data.logger.xyz
.data.logger.abc

pizi-nordic, 2018-09-14:
OK. I understood that putting section is ram is problematic and we could use it only for constant sections.

SebastianBoe, 2018-09-14:
Not sure what you mean, but anyway, the logging system is not affected by this issue, because it has ld-scripts that catch the .log_dynamic* sections so no orphan sections are created.

dcpleung, 2018-10-09:
The build system now warns about orphan sections, so developers will get hints on what might be the problem. As mentioned above, there is not generic way to forcibly place orphan sections into certain locations. So closing this...

closed 2018-10-09 by dcpleung