zephyrproject-rtos/zephyr #9506: Ztest becomes unresponsive while running SMP tests

spoorthik, 2018-08-17:
Issue: The tests in tests/kernel/smp when executed becomes unresponsive.

Sample log:

```
Running test suite smp
===================================================================
starting test - test_smp_coop_threads
PASS - test_smp_coop_threads
===================================================================
starting test - test_cpu_id_threads
```

Same set of tests when executed individually by commenting rest of the tests in testsuite like below:

```
 ztest_test_suite(smp,
/*           ztest_unit_test(test_smp_coop_threads),
*/
             ztest_unit_test(test_cpu_id_threads)
/*,
             ztest_unit_test(test_coop_resched_threads),
             ztest_unit_test(test_preempt_resched_threads),
             ztest_unit_test(test_yield_threads),
             ztest_unit_test(test_sleep_threads),
             ztest_unit_test(test_wakeup_threads),
             ztest_unit_test(test_wakeup_pending_threads)
*/
             );
    ztest_run_test_suite(smp);
```
the test case executes successfully.

Sample log:

```
Running test suite smp
===================================================================
starting test - test_cpu_id_threads
PASS - test_cpu_id_threads
===================================================================
===================================================================
PROJECT EXECUTION SUCCESSFUL
```
It is the same case for all the test cases in smp. Cleanup of threads created is handled in test case already.
```
 ztest_test_suite(smp,
/*           ztest_unit_test(test_smp_coop_threads),
             ztest_unit_test(test_cpu_id_threads),
*/
             ztest_unit_test(test_coop_resched_threads)

/*,
             ztest_unit_test(test_preempt_resched_threads),
             ztest_unit_test(test_yield_threads),
             ztest_unit_test(test_sleep_threads),
             ztest_unit_test(test_wakeup_threads),
             ztest_unit_test(test_wakeup_pending_threads)
*/
             );
    ztest_run_test_suite(smp);

Execution log:

Running test suite smp
===================================================================
starting test - test_coop_resched_threads
PASS - test_coop_resched_threads
===================================================================
===================================================================

PROJECT EXECUTION SUCCESSFUL
```

aescolar, 2018-08-19:
@spoorthik : When compiled for which target?

spoorthik, 2018-08-19:
@aescolar : Compiled for esp32.

rljordan, 2019-03-07:
@nashif Can we resolve this in future?

gnuless, 2019-03-22:
This is a can of worms that is a bit over my head at this point.  I think there are problems with races in either the xtensa code and/or the scheduler.

gnuless, 2019-03-22:
I've discovered that k_sleep() is broken on SMP, which affects this test. I'm not sure if my fix for that issue is correct, which leaves me wondering if continuing issues are created by my fix, or are actual symptoms. I have to punt this back to someone with more knowledge of the scheduler and xtensa arch.

gnuless, 2019-03-22:
FWIW, my k_sleep() fix was:

```
--- a/kernel/sched.c
+++ b/kernel/sched.c
@@ -945,6 +945,16 @@ s32_t z_impl_k_sleep(s32_t duration)
        z_remove_thread_from_ready_q(_current);
        z_add_thread_timeout(_current, ticks);

+#if defined(CONFIG_SMP)
+       /*
+        * when running SMP, we must mark the thread queued
+        * to prevent the path through z_swap() from putting
+        * it right back in a ready queue, resulting in no sleep.
+        */
+
+       z_mark_thread_as_queued(_current);
+#endif
+
        (void)z_swap(&local_lock, key);

        ticks = expected_wakeup_time - z_tick_get_32();
```

gnuless, 2019-03-27:
This bug should not have been closed, as #14846 addressed some issues related to this bug, but not all.

dleach02, 2019-12-10:
@andyross , you mentioned that you can't reproduce this anymore after #18531. Do you think we can close this issue then?

andyross, 2019-12-11:
Yes, this is working in all CI builds and has been for months. Closing.

PR #14846 (2019-03-23): kernel/sched: Mark sleeping threads suspended
On SMP, there was a bug where the logic that re-adds _current to the
run queue at swap time would accidentally reschedule threads that had
just gone to sleep, because the is_thread_prevented_from_running()
predicate only tests for threads that are "suspended" or "pending" and
not sleeping.

Overload _THREAD_SUSPENDED to indicate "sleeping" also.  Simple fix
for an immediate bug, though long term we really want to unify all the
blocked conditions to prevent this kind of state bug.

Fixes #9506
closed 2019-12-11 by andyross