zephyrproject-rtos/zephyr #9507: [PWM] No clear semantics to stop a PWM leads to diverse implementations

vaussard, 2018-08-17:
The PWM API does not mention how one can stop the timer from generating the PWM, which is critical to minimize the power consumption when going into low-power states.

The implementations of the .pin_set callback show a very diversified picture:
- STM32 and i.MX does not provide a way to completely turn off the timer
- QSMI and DW will turn off the timer by passing 'pulse_cycle = 0'
- nrf5_sw will turn off the timer by passing 'pulse_cycle = 0' or 'pulse_cycles == period_cycles' and set the GPIO to low/high respectively

The lack of a way to turn off the STM32 timer is probably a short-coming of the driver, but the diversity of the implementations make it difficult to write portable userland code.

pfalcon, 2018-08-20:
> STM32 and i.MX does not provide a way to completely turn off the timer

@erwango, @MaureenHelm : FYI

gmarull, 2020-06-15:
FYI, this PR https://github.com/zephyrproject-rtos/zephyr/pull/24569 disables the timer output when pulse_cycles is set to 0, but not the timer itself as multiple channels may be used concurrently.

henrikbrixandersen, 2020-12-30:
I think we need to distinguish between turning off the PWM output (driving it to an inactive state) and turning off the PWM controller (in order to reduce power consumption).

The former was clarified in the API documentation in 31ae13eeb16f726e56fbcfc0820675811b2f3362 and a4337bde28c8b0f79d8cc2882bbf872c90a11a9a, but some driver may not have caught up yet. The latter requires power management support in the given PWM controller driver.

carlescufi, 2021-02-16:
> The former was clarified in the API documentation in [31ae13e](https://github.com/zephyrproject-rtos/zephyr/commit/31ae13eeb16f726e56fbcfc0820675811b2f3362) and [a4337bd](https://github.com/zephyrproject-rtos/zephyr/commit/a4337bde28c8b0f79d8cc2882bbf872c90a11a9a), but some driver may not have caught up yet. The latter requires power management support in the given PWM controller driver.

Given the above I will close this now.

closed 2021-02-16 by carlescufi