zephyrproject-rtos/zephyr #9509: Unable to upload firmware over serial with mcumgr

bfrog, 2018-08-17:
Following along the guide in the samples/subsys/mgmt/mcumgr/smp_svr/README
but using --conntype=serial most of the commands work just fine except the image upload command fails.

```
mcumgr --conntype=serial --connstring=/dev/ttyACM0 image upload signed.bin  ✔  4332  10:40:58 
 0 B / 109.36 KiB [-------------------------------------------------------------------------------------------------------------------------------------------------------------]   0.00%
Error: NMP timeout
```

Is there another conntype I should be using? The docs elude to a shell conntype but I didn't see that in the docs for mcumgr

lemrey, 2018-09-05:
Hi @bfrog which board are you running this on? Could you try increasing the timeout using the `-t` switch?

parinpatel, 2018-10-16:
@bfrog are you find any solution ??

@carlescufi 
Is there any tx, rx and Hw flow control configuration is available for UART firmware upgrade??
Because I also unable to flash firmware through UART and I got following error,
Error: NMP timeout

carlescufi, 2018-10-16:
@parinpatel not that I know no. I think the config should match between the sample and mcumgr by default. But I've copied @nvlsianpu who should be able to help

carlescufi, 2018-10-16:
@parinpatel do other commands work for you? Can you try increasing the timeout with the `-t` switch?

parinpatel, 2018-10-17:
I try with -t but It didn't work 
I Have one question, for which board you were check this UART Firmware up gradation method ??
If you have mcuboot and Smp_server related README please share with me
May be I execute wrong steps

parinpatel, 2018-10-18:
Ok It's resolved now 
It happened due to It does not go to the serial recovery mode  after resolved it, I successfully upload the  firmware through UART

bfrog, 2018-10-18:
I was running this on a nrf52840 dev board, and more recently a custom board with the same chip, neither worked

parinpatel, 2018-10-19:
@bfrog 
I checked for Uart and its working as expected
which interface you are using for firmware flash UART / USB??
if UART may be I will help you for Firmware upgradation.
I was not checked for USB interface but so I haven't idea about it.

nvlsianpu, 2018-10-19:
@bfrog  An example of proper connection parameters.
`sudo ./mcumgr image list --conntype=serial --connstring='dev=/dev/ttyACM0,baud=115200'`

The mcumgr Go app doesn't use flow control (it is imposible due serial library used).

bfrog, 2018-11-09:
I finally figured this out, it had to do with the builtin jlink on the nrf52840-dk having mass storage enabled. Disabling mass storage fixes this issue! Thank you for fixing this!

closed 2018-11-09 by bfrog

gurpreetz, 2018-11-12:
Disabling mass storage on my JLINK didnt seem to help. 
This is on the nrf52dk. 

```
zephyr (dfu_expt) 15:50:37>sudo mcumgr -ldebug -t 60 --conntype=serial --connstring='dev=/dev/ttyACM0,baud=115200' image upload signed-hello1.bin
[sudo] password for gps: 
DEBU[2018-11-12 15:50:42.048] Using connection profile: name=unnamed type=serial connstring=dev=/dev/ttyACM0,baud=115200 
DEBU[2018-11-12 15:50:42.048] {add-oic-listener} [transceiver.go:64] token=[] 
DEBU[2018-11-12 15:50:42.048] {add-oic-listener} [transceiver.go:64] token=[] 
 0 / 47864 [---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------]   0.00%DEBU[2018-11-12 15:50:42.049] {add-nmp-listener} [serial_sesn.go:217] seq=66 
DEBU[2018-11-12 15:50:42.05] Encoded &{NmpBase:{hdr:{Op:2 Flags:0 Len:0 Group:1 Seq:66 Id:5}}} to:
00000000  a0                                                |.| 
DEBU[2018-11-12 15:50:42.05] Encoded:
00000000  02 00 00 01 00 01 42 05  a0                       |......B..| 
DEBU[2018-11-12 15:50:42.05] Tx NMP request: 00000000  02 00 00 01 00 01 42 05  a0                       |......B..| 
DEBU[2018-11-12 15:50:42.05] Base64 encoding request:
00000000  02 00 00 01 00 01 42 05  a0                       |......B..| 
DEBU[2018-11-12 15:50:42.05] Tx serial
00000000  06 09                                             |..| 
DEBU[2018-11-12 15:50:42.05] Tx serial
00000000  41 41 73 43 41 41 41 42  41 41 46 43 42 61 44 4d  |AAsCAAABAAFCBaDM|
00000010  6f 51 3d 3d                                       |oQ==| 
DEBU[2018-11-12 15:50:42.05] Tx serial
00000000  0a                                                |.| 
DEBU[2018-11-12 15:50:42.058] Rx serial:
00000000  06 09 41 42 41 44 41 41  41 47 41 41 46 34 65 44  |..ABADAAAGAAF4eD| 
 0 B / 46.74 KiB [---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------]   0.00%
DEBU[2018-11-12 15:51:08.947] Rx serial:                                   
 0 B / 46.74 KiB [---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------]   0.00%
```

Meanwhile over RTT on the dev board:
```
***** Booting Zephyr OS zephyr-v1.13.0-49-g121f323de *****
[MCUBOOT] [INF] main: Starting bootloader
[MCUBOOT] [INF] main: Enter the serial recovery mode
[MCUBOOT] [INF] boot_serial_output: TX
```

The gotrace:
```
mynewt.apache.org/newtmgr/vendor/mynewt.apache.org/newt/util.NewNewtError(0xb6bd2f, 0xb, 0xb)
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/vendor/mynewt.apache.org/newt/util/util.go:75 +0xd6
mynewt.apache.org/newtmgr/vendor/mynewt.apache.org/newt/util.ChildNewtError(0xc0a440, 0xc000068ed0, 0xc00020c000)
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/vendor/mynewt.apache.org/newt/util/util.go:101 +0x85
mynewt.apache.org/newtmgr/newtmgr/cli.imageUploadCmd(0xc0001c78c0, 0xc000092600, 0x1, 0x6)
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/newtmgr/cli/image.go:203 +0x3a4
mynewt.apache.org/newtmgr/vendor/github.com/spf13/cobra.(*Command).execute(0xc0001c78c0, 0xc0001caaa0, 0x6, 0xa, 0xc0001c78c0, 0xc0001caaa0)
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/vendor/github.com/spf13/cobra/command.go:654 +0x2af
mynewt.apache.org/newtmgr/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0xc0001c6000, 0xc0001c6480, 0xc0001c6240, 0xc0001580e0)
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/vendor/github.com/spf13/cobra/command.go:729 +0x2cc
mynewt.apache.org/newtmgr/vendor/github.com/spf13/cobra.(*Command).Execute(0xc0001c6000, 0xb90c78, 0xc000150360)
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/vendor/github.com/spf13/cobra/command.go:688 +0x2b
main.main()
	/home/gps/src/go/src/github.com/apache/mynewt-mcumgr-cli/mcumgr/mcumgr.go:94 +0x10d

goroutine 19 [syscall]:
os/signal.signal_recv(0x0)
	/home/gps/src/golang/go/src/runtime/sigqueue.go:139 +0x9c
os/signal.loop()
	/home/gps/src/golang/go/src/os/signal/signal_unix.go:23 +0x22
created by os/signal.init.0
	/home/gps/src/golang/go/src/os/signal/signal_unix.go:29 +0x41

goroutine 34 [chan receive]:
main.main.func1(0xc000150360)
	/home/gps/src/go/src/github.com/apache/mynewt-mcumgr-cli/mcumgr/mcumgr.go:82 +0x41
created by main.main
	/home/gps/src/go/src/github.com/apache/mynewt-mcumgr-cli/mcumgr/mcumgr.go:80 +0x103

[... 24 lines cut ...]
goroutine 36 [select]:
mynewt.apache.org/newtmgr/nmxact/omp.(*Dispatcher).addOmpListener.func1(0xc000168480, 0xc0001564c0)
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/nmxact/omp/dispatch.go:77 +0x17c
created by mynewt.apache.org/newtmgr/nmxact/omp.(*Dispatcher).addOmpListener
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/nmxact/omp/dispatch.go:72 +0xb6

goroutine 37 [select]:
mynewt.apache.org/newtmgr/nmxact/omp.(*Dispatcher).addOmpListener.func1(0xc000168500, 0xc0001565c0)
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/nmxact/omp/dispatch.go:77 +0x17c
created by mynewt.apache.org/newtmgr/nmxact/omp.(*Dispatcher).addOmpListener
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/nmxact/omp/dispatch.go:72 +0xb6

[... cut ...]
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/nmxact/nmserial/serial_sesn.go:101 +0x261

goroutine 39 [select]:
mynewt.apache.org/newtmgr/vendor/github.com/cheggaaa/pb.(*ProgressBar).refresher(0xc000212000)
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/vendor/github.com/cheggaaa/pb/pb.go:462 +0xce
created by mynewt.apache.org/newtmgr/vendor/github.com/cheggaaa/pb.(*ProgressBar).Start
	/home/gps/src/go/src/mynewt.apache.org/newtmgr/vendor/github.com/cheggaaa/pb/pb.go:123 +0xb7 
Error: NMP timeout
```

YuraCobain, 2019-07-11:
Hello Gurpreet,

A little bit out of date but I had a similar issue that was caused by opening the same serial device by different processed during upgrade.
In my case it was minicom and mcumgr that simutenisly access /dev/ttyUSB0.
Make sure that serial is accessed only by mcumgr solved the issue.

BR,
Iurii

nvlsianpu, 2019-07-11:
@gurpreetz From what I can see in your log communication doesn't happened on you setup, so probably above advice is the best.

mayankpatel17, 2019-11-11:
@parinpatel I'm using mcuboot bootloader in serial recovery mode and smp_svr application for DFU over UART. I can't see 'mcumgr' command respond for even single time. Every time i'm getting NMP timeout error. It seems like i am missing something important while building the 'mcuboot' & 'smp_svr'.

Could you share the README for the correct/valid config for smp_svr (prj.conf).

parinpatel, 2019-11-12:
@mayankpatel17 
Hi Mayank,
In my case, It happened because It does not go to the serial recovery mode after resolved it, I successfully upload the firmware through UART

**Can you validate that your device enters into the serial recovery mode**?

mayankpatel17, 2019-11-12:
@parinpatel 
Now after upgrading my zephyr version to 2.1.0 (Using west update), I'm able to use mcumgr command. I have also uploaded sample hci_uart image using (mcumgr image upload) command.

I have one more question, when i flash mcuboot + zephyr application on Nordic Nrf52840_pca10056 at the same time (Both have different address space). Application is not working in that case.

Thanks,
Mayank

parinpatel, 2019-11-12:
Hi Mayank,
Yes, bootloader and application have a different address.
In this case, on a startup, a program first jumps to the bootloader address and then finally jump to the application address.
Please use the sample application which has combine include bootloader+application.
NOTE: Please start with sample hello word or led blinking application (combine bootloader + application) which is already available in zephyr sdk
and Also see the error logs.

ya-mouse, 2021-01-05:
Just in case. I got zero progress because the firmware image was bigger than flash area (defined in DTS). It would be great if mcuboot will return appropriate return code if data wouldn't fit the flash partition.