zephyrproject-rtos/zephyr #9523: tests/kernel/mem_protect/stackprot hangs without any console output on nrf51/52

pswarnak, 2018-08-20:
The test **tests/kernel/mem_protect/stackprot** hangs without any console log or error message.

**Arch:** arm
**Board:** nrf51/52 - nrf51_pca10028:arm, nrf52_pca10040:arm, nrf52840_pca10056:arm
**Zephyr commit:** f3bc967e499b6dd252c7b407c7019460a66a1174
**Toolchain:**  ZEPHYR_TOOLCHAIN_VARIANT=zephyr

**Error:**
*No console log available at all.*

**Steps to reproduce:**
```
cd zephyr/tests/kernel/mem_protect/stackprot
rm -rf build && mkdir build && cd build
cmake -D BOARD=nrf52_pca10040 ../
make BOARD=nrf52_pca10040 flash

Check the console log.
```

**Sample Expected output:**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.12.0-1543-gf3bc967 (delayed boot 1000ms) *****
Starts alternate_thread
alternate_thread: Input string is too long and stack overflowed!

***** Stack Check Fail! *****
Current thread ID = 0x20000030
Faulting instruction address = 0x78ec
Fatal fault in thread 0x20000030! Aborting.
Running test suite stackprot
===================================================================
starting test - test_stackprot
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
PASS - test_stackprot
===================================================================
===================================================================
RunID: :pa7o
PROJECT EXECUTION SUCCESSFUL
```

nashif, 2018-08-21:
@pizi-nordic can you take a look?

pswarnak, 2018-08-31:
@pizi-nordic:
FYI: **tests/kernel/mem_protect/stack_random** also hangs without any console output on nrf52840_pca10056. Kindly check.

oliviermartin, 2018-08-31:
I had to disable stack protection to make my Zephyr nRF52840-based application running on v1.13.0. I would suggest to increase the priority of the issue.

nashif, 2018-08-31:
related to CONFIG_STACK_CANARIES...

carlescufi, 2018-09-03:
@nashif since the PR from @pizi-nordic is a bit complex, @cvinayak will now send a simpler fix for the upcoming release. We can the revisit Piotr's PR later.

PR #9770 (2018-09-03): drivers: entropy: nrf5: Fix ENTROPY_BUSYWAIT from hanging
Missing clear pending RNG_IRQn when generating multiple
random numbers caused CPU to hang at __WFE call in the
loop waiting for new value.

This fixes commit ddb7f88f9ed5 ("drivers: entropy: nrf5: Fix
ENTROPY_BUSYWAIT implementation")

Fixes #9523.
commit 1359992d (2018-09-03): drivers: entropy: nrf5: Fix ENTROPY_BUSYWAIT from hanging

Missing clear pending RNG_IRQn when generating multiple
random numbers caused CPU to hang at __WFE call in the
loop waiting for new value.

This fixes commit ddb7f88f9ed5 ("drivers: entropy: nrf5: Fix
ENTROPY_BUSYWAIT implementation")

Fixes #9523.
closed 2018-09-03 by carlescufi