zephyrproject-rtos/zephyr #9530: Bluetooth/gatt: bt_gatt_notify never return -ENOMEM, undocumented return value.

KAGA164, 2018-08-20:
The bt_gatt_notify function calls gatt_notify, which should return -ENOMEM if buffer for pdu can not be allocated, but this situation will never occur because the buffer allocation function is called from the K_FOREVER timeout, so the thread will wait for the buffer memory allocation and will never return NULL value which would be a reason to return -ENOMEM.

jhedberg, 2018-08-23:
So do I understand right that you want the following change:
```
--- a/subsys/bluetooth/host/gatt.c
+++ b/subsys/bluetooth/host/gatt.c
@@ -669,10 +669,6 @@ static int gatt_notify(struct bt_conn *conn, u16_t handle, const void *data,
        struct bt_att_notify *nfy;
 
        buf = bt_att_create_pdu(conn, BT_ATT_OP_NOTIFY, sizeof(*nfy) + len);
-       if (!buf) {
-               BT_WARN("No buffer available to send notification");
-               return -ENOMEM;
-       }
 
        BT_DBG("conn %p handle 0x%04x", conn, handle);
 
```

Do you want to submit a PR for it yourself?

KAGA164, 2018-08-23:
Yes, I think checking this value can be a bit confusing. Ok, so I'll do pull-request myself.

closed 2018-08-23 by KAGA164