zephyrproject-rtos/zephyr #9535: broken callback handling in nrfx gpio driver

pdunaj, 2018-08-20:
In `drivers/gpio/gpio_nrfx.c` function `gpiote_pin_int_cfg` calls `gpiote_channel_alloc` and takes its return value as its own.
The problem is that `gpiote_channel_alloc` (if successful) returns the channel number. In this case value is interpreted as an error.

I guess the best fix is to return 0 in `gpiote_channel_alloc` every time channel is found.

The problem is that new gpio driver seems have more bugs. When I disable callbacks while in callback, I receive another interrupt. It was working correctly and disabling the callback as expected in the old driver.

pdunaj, 2018-08-20:
Fix is ready - I will try to push it in few minutes

pdunaj, 2018-08-20:
@carlescufi Can I push the change to our fork or should it go to main zephyr repo? If it is the second case, will you pull it to our fork from there?

carlescufi, 2018-08-20:
@pdunaj let's fix it upstream

PR #9538 (2018-08-27): drivers: gpio: Fix two bugs in nrfx gpio
Commit fixes two issues:
- Return correct value from gpiote_channel_alloc
- Clear interupt event when callback is disabled

Jira:DESK-270

Fixes #9535
commit fb10377e (2018-08-27): drivers: gpio: Fix two bugs in nrfx gpio

Commit fixes two issues:
- Return correct value from gpiote_channel_alloc
- Check if pending event interrupt was not disabled

Jira:DESK-270

Fixes #9535
closed 2018-08-27 by nashif