zephyrproject-rtos/zephyr #9536: console: missing kernel.h include in header

lemrey, 2018-08-20:
console.h uses k_fifo but kernel.h is not included.

pfalcon, 2018-08-20:
@lemrey : How did you hit it? (Generally, we should back such things with regression tests. Maybe we'll be too lazy to write one, but we should not be lazy to discuss *how* to write one ;-).

lemrey, 2018-08-20:
I hit a compiler warning; perhaps it is possible to reproduce by including console.h before kernel.h:
`warning: ‘struct k_fifo’ declared inside parameter list will not be visible outside of this definition or declaration`

pfalcon, 2018-08-20:
Thanks. Fairly speaking, I had an idea that all Zephyr apps should start with `#include <zephyr.h>`. Dunno where I got the from though, probably because all in-tree samples do that. Anyway, it's an easy fix, posted as https://github.com/zephyrproject-rtos/zephyr/pull/9541

PR #9541 (2018-08-20): include: console: Include kernel.h for struct k_fifo
console.h references struct k_fifo for an argument, so include header
where it's defined.

Fixes: #9536
commit 96784dff (2018-08-20): include: console: Include kernel.h for struct k_fifo

console.h references struct k_fifo for an argument, so include header
where it's defined.

Fixes: #9536
closed 2018-08-20 by nashif