zephyrproject-rtos/zephyr #9537: ENC28J60 can‘t receive packets properly

lgl88911, 2018-08-20:
After the initialization of the ENC28J60, only the first one was correct when the packet was received. I found  the ERDPT small than ERXRDPT, after received first packet.  It casue RX FIFO read error.
Does anybody have a problem like that? If yes, I have a patch for the issue, may be pr.
Thanks!

pfalcon, 2018-08-21:
> only the first one was correct when the packet was received.

First one what? Can you please describe the problem in more detail, perhaps giving steps how to reproduce it.

lgl88911, 2018-08-22:
@pfalcon 
Reproduce the issue 100%, receive packet error  will cause ipstack can't work, There is an easy way to reproduce issue:
1. Application code and configuration file at
https://gitee.com/lgl88911/nrf52_moderate/blob/master/apps/basetest/src/nettest.c
https://gitee.com/lgl88911/nrf52_moderate/blob/master/boards/arm/nrf52_moderate/nrf52_moderate_defconfig
https://gitee.com/lgl88911/nrf52_moderate/blob/master/apps/basetest/prj.conf

2. Power on the board & wait receive net packet.
3. Send 2 packets form PC(test.cap total 42 byte):
    tcpreplay -i interface -p 2 test.cap
4. check rev packet param in eth_enc28j60_rx :
The first pack is correct:
    next_packet is 70
    frm_len is 60
    diff 10 byte is(2 next packet+4 byte status+4byte crc)
but The second packet is wrong, cause get error data:
    next_packet is 31244
    frn_len is more than 1500
the correct value of second should be:
    next_packet is 140
    frm_len is 60

5. debug:
    Before read second packet, check reg ERDPT&ERXRDPT, ERDPT is smaller than ERXRDPT 4. 
    I think ERDPT is incorrect, which cause read data dislocation.

PR #9616 (2018-08-31): Fixed #9537 enc28j60 rev error
drivers: eth: enc28j60: Fix read error when ERDPT < ERXRDPT

After enc28j60 has read first packet and if ERDPT is smaller than ERXRDPT, then this will cause read Rx FIFO error.
The fix is to set ERDPT properly before reading next packet.

Fixed #9537
commit 49343239 (2018-08-31): drivers: eth: enc28j60: Fix read error when ERDPT < ERXRDPT

After read first packet and if ERDPT < ERXRDPT,cause read Rx FIFO error.
The fix is to set ERDPT properly before reading next packet.

Fixed #9537
closed 2018-08-31 by nashif