zephyrproject-rtos/zephyr #9558: Building examples with UART output for FRDM_K64F defunct

therealprof, 2018-08-22:
When trying to build examples (e.g. samples/net/lwm2m_client) with UART output for my FRDM_K64F I get the following compiler errors:
```
/Users/egger/oss/zephyr/drivers/serial/uart_mcux.c: In function 'uart_mcux_config_func_0':
/Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:312:1: error: macro "_ARCH_IRQ_CONNECT" passed 7 arguments, but takes just 5
       uart_mcux_isr, DEVICE_GET(uart_0), 0);
 ^     ~~~~~~~~~
In file included from /Users/egger/oss/zephyr/include/arch/arm/cortex_m/irq.h:17:0,
                 from /Users/egger/oss/zephyr/include/arch/arm/arch.h:27,
                 from /Users/egger/oss/zephyr/include/arch/cpu.h:15,
                 from /Users/egger/oss/zephyr/include/kernel_includes.h:34,
                 from /Users/egger/oss/zephyr/include/kernel.h:17,
                 from /Users/egger/oss/zephyr/include/device.h:11,
                 from /Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:8:
/Users/egger/oss/zephyr/include/irq.h:50:2: error: '_ARCH_IRQ_CONNECT' undeclared (first use in this function)
  _ARCH_IRQ_CONNECT(irq_p, priority_p, isr_p, isr_param_p, flags_p)
  ^
/Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:310:2: note: in expansion of macro 'IRQ_CONNECT'
  IRQ_CONNECT(CONFIG_UART_MCUX_0_IRQ_STATUS,
  ^~~~~~~~~~~
/Users/egger/oss/zephyr/include/irq.h:50:2: note: each undeclared identifier is reported only once for each function it appears in
  _ARCH_IRQ_CONNECT(irq_p, priority_p, isr_p, isr_param_p, flags_p)
  ^
/Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:310:2: note: in expansion of macro 'IRQ_CONNECT'
  IRQ_CONNECT(CONFIG_UART_MCUX_0_IRQ_STATUS,
  ^~~~~~~~~~~
/Users/egger/oss/zephyr/samples/net/lwm2m_client/test/zephyr/include/generated/generated_dts_board.h:374:46: error: expected expression before '[' token
 #define NXP_KINETIS_UART_4006A000_IRQ_0      [31, 0]
                                              ^
/Users/egger/oss/zephyr/include/irq.h:230:42: note: in definition of macro 'irq_enable'
 #define irq_enable(irq) _arch_irq_enable(irq)
                                          ^~~
/Users/egger/oss/zephyr/samples/net/lwm2m_client/test/zephyr/include/generated/generated_dts_board.h:385:50: note: in expansion of macro 'NXP_KINETIS_UART_4006A000_IRQ_0'
 #define NXP_KINETIS_UART_4006A000_IRQ_STATUS     NXP_KINETIS_UART_4006A000_IRQ_0
                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/egger/oss/zephyr/samples/net/lwm2m_client/test/zephyr/include/generated/generated_dts_board.h:429:40: note: in expansion of macro 'NXP_KINETIS_UART_4006A000_IRQ_STATUS'
 #define CONFIG_UART_MCUX_0_IRQ_STATUS  NXP_KINETIS_UART_4006A000_IRQ_STATUS
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:314:13: note: in expansion of macro 'CONFIG_UART_MCUX_0_IRQ_STATUS'
  irq_enable(CONFIG_UART_MCUX_0_IRQ_STATUS);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/egger/oss/zephyr/samples/net/lwm2m_client/test/zephyr/include/generated/generated_dts_board.h:374:49: warning: left-hand operand of comma expression has no effect [-Wunused-value]
 #define NXP_KINETIS_UART_4006A000_IRQ_0      [31, 0]
                                                 ^
/Users/egger/oss/zephyr/include/irq.h:230:42: note: in definition of macro 'irq_enable'
 #define irq_enable(irq) _arch_irq_enable(irq)
                                          ^~~
/Users/egger/oss/zephyr/samples/net/lwm2m_client/test/zephyr/include/generated/generated_dts_board.h:385:50: note: in expansion of macro 'NXP_KINETIS_UART_4006A000_IRQ_0'
 #define NXP_KINETIS_UART_4006A000_IRQ_STATUS     NXP_KINETIS_UART_4006A000_IRQ_0
                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/egger/oss/zephyr/samples/net/lwm2m_client/test/zephyr/include/generated/generated_dts_board.h:429:40: note: in expansion of macro 'NXP_KINETIS_UART_4006A000_IRQ_STATUS'
 #define CONFIG_UART_MCUX_0_IRQ_STATUS  NXP_KINETIS_UART_4006A000_IRQ_STATUS
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:314:13: note: in expansion of macro 'CONFIG_UART_MCUX_0_IRQ_STATUS'
  irq_enable(CONFIG_UART_MCUX_0_IRQ_STATUS);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/egger/oss/zephyr/include/arch/arm/cortex_m/irq.h:18:0,
                 from /Users/egger/oss/zephyr/include/arch/arm/arch.h:27,
                 from /Users/egger/oss/zephyr/include/arch/cpu.h:15,
                 from /Users/egger/oss/zephyr/include/kernel_includes.h:34,
                 from /Users/egger/oss/zephyr/include/kernel.h:17,
                 from /Users/egger/oss/zephyr/include/device.h:11,
                 from /Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:8:
/Users/egger/oss/zephyr/samples/net/lwm2m_client/test/zephyr/include/generated/generated_dts_board.h:427:39: error: 'NXP_KINETIS_UART_4006A000_IRQ_ERROR' undeclared (first use in this function); did you mean 'NXP_KINETIS_UART_4006A000_IRQ_0'?
 #define CONFIG_UART_MCUX_0_IRQ_ERROR  NXP_KINETIS_UART_4006A000_IRQ_ERROR
                                       ^
/Users/egger/oss/zephyr/include/sw_isr_table.h:70:5: note: in definition of macro '_ISR_DECLARE'
    {irq, flags, &func, (void *)param}
     ^~~
/Users/egger/oss/zephyr/include/irq.h:50:2: note: in expansion of macro '_ARCH_IRQ_CONNECT'
  _ARCH_IRQ_CONNECT(irq_p, priority_p, isr_p, isr_param_p, flags_p)
  ^~~~~~~~~~~~~~~~~
/Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:316:2: note: in expansion of macro 'IRQ_CONNECT'
  IRQ_CONNECT(CONFIG_UART_MCUX_0_IRQ_ERROR,
  ^~~~~~~~~~~
/Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:316:14: note: in expansion of macro 'CONFIG_UART_MCUX_0_IRQ_ERROR'
  IRQ_CONNECT(CONFIG_UART_MCUX_0_IRQ_ERROR,
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/egger/oss/zephyr/include/arch/arm/arch.h:27:0,
                 from /Users/egger/oss/zephyr/include/arch/cpu.h:15,
                 from /Users/egger/oss/zephyr/include/kernel_includes.h:34,
                 from /Users/egger/oss/zephyr/include/kernel.h:17,
                 from /Users/egger/oss/zephyr/include/device.h:11,
                 from /Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:8:
/Users/egger/oss/zephyr/samples/net/lwm2m_client/test/zephyr/include/generated/generated_dts_board.h:428:42: error: 'NXP_KINETIS_UART_4006A000_IRQ_ERROR_PRIORITY' undeclared (first use in this function); did you mean 'NXP_KINETIS_UART_4006A000_IRQ_0_PRIORITY'?
 #define CONFIG_UART_MCUX_0_IRQ_ERROR_PRI NXP_KINETIS_UART_4006A000_IRQ_ERROR_PRIORITY
                                          ^
/Users/egger/oss/zephyr/include/arch/arm/cortex_m/irq.h:84:27: note: in definition of macro '_ARCH_IRQ_CONNECT'
  _irq_priority_set(irq_p, priority_p, flags_p); \
                           ^~~~~~~~~~
/Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:316:2: note: in expansion of macro 'IRQ_CONNECT'
  IRQ_CONNECT(CONFIG_UART_MCUX_0_IRQ_ERROR,
  ^~~~~~~~~~~
/Users/egger/oss/zephyr/drivers/serial/uart_mcux.c:317:7: note: in expansion of macro 'CONFIG_UART_MCUX_0_IRQ_ERROR_PRI'
       CONFIG_UART_MCUX_0_IRQ_ERROR_PRI,
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_mcux.c.obj] Error 1
make[1]: *** [zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/all] Error 2
make: *** [all] Error 2
```

therealprof, 2018-08-22:
I just went back a couple of months (41d5808321acdc0ac936b8da342ba00521e30d4d) to figure where it broke but couldn't find any culprit so this is most likely caused by a change in environment. Other builds (e.g. nucleo_f429zi) work just fine.

pfalcon, 2018-08-22:
@therealprof: I can't reproduce this with master of 49732b27d939eab71eeef317c76135b1ca1af51a (HEAD of the moment):

~~~
[ 98%] Building C object zephyr/CMakeFiles/kernel_elf.dir/isr_tables.c.obj
[100%] Linking C executable zephyr.elf
Generating files from zephyr.elf for board: frdm_k64f
[100%] Built target kernel_elf
pfalcon@x230:~/projects-3rdparty/Embedded/Zephyr/zephyr/samples/net/lwm2m_client$
~~~

It's usual Zephyr SDK 0.9.3 which didn't change in a while, etc. Did you try to "rm -rf" the output dir? When did it build last time for you, anything might have changed on your side since that?

pfalcon, 2018-08-22:
> drivers/serial/uart_mcux.c:312:1: error: macro "_ARCH_IRQ_CONNECT" passed 7 arguments, but takes just 5

If master, that line has 5 arguments: https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/serial/uart_mcux.c#L312

Any chance that you're on another branch or played with some patches left in your tree?

pfalcon, 2018-08-22:
For reference, I have `Version: DTC 1.4.0`.

And yes, looks like it might be related, based on warnings like:

> /Users/egger/oss/zephyr/samples/net/lwm2m_client/test/zephyr/include/generated/generated_dts_board.h:374:49: warning: left-hand operand of comma expression has no effect [-Wunused-value]
 #define NXP_KINETIS_UART_4006A000_IRQ_0      [31, 0]

Can you try to downgrade and see if it helps?

therealprof, 2018-08-22:
@pfalcon Bingo. DTC 1.4.0 works.

I can see a number of changes in the generated files, e.g. the compiled DTS changed here:
```
-                       interrupts = <0x1f 0x0 0x20 0x0>;
+                       interrupts = < 0x1f 0x00 >, < 0x20 0x00 >;
```

Why this only seems to affect the NXP K series is a mystery to me. I tried various builds, including `bbc_microbit`, various `nucleo` and even `frdm_kw41z` with the new `dtc` but only `frdm_k64f` fails.

pfalcon, 2018-08-22:
Weird indeed. Well, I guess we need to re-brand it as DTC issue then (I'm making quick changes, feel free to adjust as needed) and let device tree folks know.

pfalcon, 2018-08-22:
@agross-linaro : Assigning to you by old memory. Please reassign if needed.

@erwango, @nashif : FYI too.

PR #9805 (2018-09-06): scripts: dts: extract: Fix handling of interrupt prop being a list of…
… lists

Before dtc 1.4.7 we'd get something like the following for an interrupt
property in which #interrupt-cells is 2.

interrupts = <1 2 3 4>;

After dtc 1.4.7 we get:

interrupts = <1 2>, <3 4>;

We should handle both cases in the extract interrupt handling code.  So
if we see a list of lists, we flatten it to a single list to normalize
the property.

Fixes: #9558
commit 5824b811 (2018-09-06): scripts: dts: extract: Fix handling of irq prop being a list of lists

Before dtc 1.4.7 we'd get something like the following for an interrupt
property in which #interrupt-cells is 2.

interrupts = <1 2 3 4>;

After dtc 1.4.7 we get:

interrupts = <1 2>, <3 4>;

We should handle both cases in the extract interrupt handling code.  So
if we see a list of lists, we flatten it to a single list to normalize
the property.

Fixes: #9558
closed 2018-09-06 by galak