zephyrproject-rtos/zephyr #9566: Unclear definition of CONFIG_IS_BOOTLOADER

JoeHut, 2018-08-22:
The ``CONFIG_IS_BOOTLOADER`` option was apparently introduced by STM for vector relocation on Cortex-M0. The naming or its usage is misleading since the config's help says "Zephyr will act as a bootloader to execute" but the config is selected in board files. What I would expect that this option is only used in applications that act as a bootloader such as MCUboot.

nvlsianpu, 2018-08-22:
CONFIG_IS_BOOTLOADER used to enable vector relocation mechanism which is specific for cortex-m0 stm32f0 family. Parallel to this feature some time ago we implemented generic cortex-m0 vector relocation (https://github.com/zephyrproject-rtos/zephyr/pull/4536/files). What I can suggest is to rename this keyword to something more suitable. I think feature enabled by this keyword is something vendor specific.

nvlsianpu, 2018-08-22:
cc @erwango If you can answer in your convenience or redirect to someone (no rush).

b0661, 2018-08-23:
CONFIG_IS_BOOTLOADER is used for the situation where you **do not need vector relocation** for the cortex-m0.

SebastianBoe, 2018-08-23:
This issue relates to: https://github.com/zephyrproject-rtos/zephyr/issues/6180

nvlsianpu, 2018-08-24:
> CONFIG_IS_BOOTLOADER is used for the situation where you do not need vector relocation for the cortex-m0.

Ach yes, you are right.

erwango, 2019-02-15:
CONFIG_IS_BOOTLOADER was introduced by 21628d41d02115090208b42739ff5448c801900a so not introduced by or for STM.
Since authors do not contribute to zephyr anymore, maybe @andyross would be the right person for this point?

erwango, 2020-05-25:
@JoeHut Is this issue still valid ?

JoeHut, 2020-05-25:
We worked around it and I am currently not up to date with the 2.x branches since we stayed with the LTS. If noone else complains I guess you can close it.

erwango, 2020-05-25:
Txs @JoeHut , I'm closing

closed 2020-05-25 by erwango