zephyrproject-rtos/zephyr #9570: Network stack cleanup: real TCP cleanup

pfalcon, 2018-08-22:
This is continuation of https://github.com/zephyrproject-rtos/zephyr/issues/8726. While that touches surface things, this ticket proposes to go far deeper. Overall, the TCP code is one of the most convoluted and buggy pieces of the IP stack.

[ ] NET_CONN_CB considered harmful: https://github.com/zephyrproject-rtos/zephyr/issues/8723#issuecomment-415040335
[ ] Moreover, at 3 peices, we likely have too many such connection callbacks. They can be seen as optimization of implementing TCP state machine, but that's definitely a premature optimization, leading to convolution, duplication, inconsistency. TCP defines one state machine, so that means one connection callback (implementing entire state machine). Maybe, just maybe it would make sense to optimize that to 2 callbacks: for active vs listening sockets. Anything else is truly hardcode optimization, which should be performed on fully working and validated stack.
[ ] A large part of UDP vs TCP send path still lives in net_context.c:sendto(), having a spaghetti code trying to deal with both at once. No need, the are quite different. That func should do validation of addresses, and both protocols should part their ways into udp.c and tcp.c ASAP after that.
[ ] FIN is not queued: https://github.com/zephyrproject-rtos/zephyr/issues/8188
[ ] Likewise, SYN is not queued.
[ ] Much, much more...

jukkar, 2018-08-23:
Pointless to create a new PR for this, just place the stuff into #8726

pfalcon, 2018-08-23:
> Pointless to create a new PR for this, just place the stuff into #8726

Ticket you mean? Well, you see, we can't even agree whether it makes sense to create a new ticket, just imagine how we can disagree about content of tickets, putting too much different things into one may make it harder for everyone ;-). Seriously, things in #8726 look "decided" (and at the same time I lack info on exact motivation of array -> slist global refactor), whereas things in this list may need more detailed investigation and consideration. I didn't want to break fun to whoever will work on #8726 of being close it in reasonable time (and risk my points being closed with it, which happens too).

jukkar, 2018-08-23:
Even the title of these issues is same (minus the "real" word on this one). I mean seriously, just use the older ticket for this, there really no need to create a new one.

pfalcon, 2018-08-23:
Well, thanks for feedback, will do it like that next time. But now this ticket is here, let it be I guess. Actually, I wanted to split comments from https://github.com/zephyrproject-rtos/zephyr/issues/8723#issuecomment-415040335 into separate ticket too, as it looks like it may need good investigation. What do you think?

pfalcon, 2018-08-23:
I agree, but "lowering" should hopefully happen by discussion and resolving, not non-submitting. Anyway, looking forward for your comments on my tickets, and let me know if you expect feedback on some of your tickets you'd like to resolve soon.

pfalcon, 2018-08-23:
> We have tcp->flags NET_TCP_FINAL_SENT. Guess what - it's unused. That's because there're standard TCP states instead.

We actually have NET_TCP_FINAL_RECV too. So, we have two parallel pairs of tcp->flags NET_TCP_FINAL_SENT/NET_TCP_FINAL_RECV, and  tcp->fin_sent / tcp->fin_rcvd. Any developer can choose what they like. Or create 3rd pair.

tbursztyka, 2019-04-17:
Closing this, legacy TCP is not going to be cleaned up.

TCP is being reworked here #11443

closed 2019-04-17 by tbursztyka