zephyrproject-rtos/zephyr #9582: Cannot find g++ when CONFIG_CPLUSPLUS is set to y

zfviwx, 2018-08-23:
Because it is missing ".exe" in the path when setting ${cplusplus_compiler}

SebastianBoe, 2018-08-23:
Would you be able to test this patch?

```
diff --git a/cmake/compiler/gcc.cmake b/cmake/compiler/gcc.cmake
index d301a7754d..ca9031c24b 100644
--- a/cmake/compiler/gcc.cmake
+++ b/cmake/compiler/gcc.cmake
@@ -49,7 +49,7 @@ else()
     set(cplusplus_compiler ${CMAKE_C_COMPILER})
   endif()
 endif()
-set(CMAKE_CXX_COMPILER ${cplusplus_compiler}     CACHE INTERNAL " " FORCE)
+find_program(CMAKE_CXX_COMPILER ${cplusplus_compiler} PATH ${TOOLCHAIN_HOME} NO_DEFAULT_PATH)
 
 set(NOSTDINC "")

```

zfviwx, 2018-08-23:
This solves the problem, and also would be my suggestion.

PR #9586 (2018-09-12): cmake: Use find_program on CMAKE_CXX_COMPILER
'find_program' should be used instead of 'set' for determining the
paths of host programs. It resolves #9582 and various other issues.
commit 1d90bcdc (2018-09-12): cmake: Use find_program on CMAKE_CXX_COMPILER

'find_program' should be used instead of 'set' for determining the
paths of host programs. It resolves #9582 and various other issues.
closed 2018-09-12 by nashif