zephyrproject-rtos/zephyr #9627: Clean up timeout handling

andyross, 2018-08-25:
See discussion in #9620  and #8669 

The handle_expired_timeouts() mechanism is fragile.  The specific bug was patched, but the code is too large and complicated for what it does (trying to build a list of expired timeouts ahead of time before invoking them instead of just doing a while loop on the queue with simpler constraints). 

It also needs some work to unify the tickless handling with the normal timeout stuff, which is needlessly divergent.

findlayfeng, 2018-08-30:
Here you need to keep next.
Because the current timeout will destroy the pointer to next if it is re-added, continuing to follow the next loop will have unpredictable results.
Just need to remove remove

andyross, 2018-11-15:
Timeouts got reworked in the timer rewrite.  Closing my own bug, because obviously I think my new implementation is better than the old one.

closed 2018-11-15 by andyross