zephyrproject-rtos/zephyr #9630: STM32L4: cannot use more than one pin with interrupt on a given port

kzyapkov, 2018-08-25:
This code reproduces the issue:

```
#include <zephyr.h>
#include <gpio.h>
#include <misc/printk.h>

static struct gpio_callback btn_gpioa_cb;
static struct gpio_callback btn_gpiob_cb;

void _on_button(struct device *gpio, struct gpio_callback *cb,
		u32_t pins)
{
	printk("p %p cb %p pins %08x\n", gpio, cb, pins);
}

void _init_buttons(void)
{
	struct device *gpioa = device_get_binding("GPIOA");
	struct device *gpiob = device_get_binding("GPIOB");

	gpio_pin_configure(gpioa, 10,
			   (GPIO_DIR_IN | GPIO_INT | GPIO_INT_EDGE |
			    GPIO_INT_DOUBLE_EDGE | GPIO_PUD_PULL_UP));

	gpio_pin_configure(gpiob, 4,
			   (GPIO_DIR_IN | GPIO_INT | GPIO_INT_EDGE |
			    GPIO_INT_DOUBLE_EDGE | GPIO_PUD_PULL_UP));

	gpio_pin_configure(gpiob, 5,
			   (GPIO_DIR_IN | GPIO_INT | GPIO_INT_EDGE |
			    GPIO_INT_DOUBLE_EDGE | GPIO_PUD_PULL_UP));

	gpio_init_callback(&btn_gpioa_cb, _on_button, BIT(10));
	gpio_init_callback(&btn_gpiob_cb, _on_button, BIT(4) | BIT(5));

	if (gpio_add_callback(gpioa, &btn_gpioa_cb) < 0) {
		printk("btn cb failed %s", "GPIOA");
		return;
	}

	if (gpio_add_callback(gpiob, &btn_gpiob_cb) < 0) {
		printk("btn cb failed %s", "GPIOB");
		return;
	}

	gpio_pin_enable_callback(gpioa, 10);
	gpio_pin_enable_callback(gpiob, 4);
	gpio_pin_enable_callback(gpiob, 5);
}

void main(void)
{
	_init_buttons();

	while (true) {
		k_sleep(K_SECONDS(1));
		printk(".");
	}
}
```

This was tested on a Nucleo L476RG. Events on GPIOA10 and GPIOB5 are detected, but GPIOB4 is silent. It doesn't matter if individual `struct gpio_callback`-s are used per pin or per port.

ydamigos, 2018-08-29:
```
diff --git a/arch/arm/soc/st_stm32/stm32l4/soc_gpio.c b/arch/arm/soc/st_stm32/stm32l4/soc_gpio.c
index 1856eea70..60a202cac 100644
--- a/arch/arm/soc/st_stm32/stm32l4/soc_gpio.c
+++ b/arch/arm/soc/st_stm32/stm32l4/soc_gpio.c
@@ -156,7 +156,7 @@ int stm32_gpio_enable_int(int port, int pin)
                return -EINVAL;
        }
 
-       *reg &= STM32L4X_SYSCFG_EXTICR_PIN_MASK << ((pin % 4) * 4);
+       *reg &= ~(STM32L4X_SYSCFG_EXTICR_PIN_MASK << ((pin % 4) * 4));
        *reg |= port << ((pin % 4) * 4);
 
        return 0; /* Nothing to do here for STM32L4s */
```

@kzyapkov Could you try the above patch?

kzyapkov, 2018-08-29:
@ydamigos the patch fixed it for me, tested a few combinations of ports and pins.

PR #9715 (2018-08-29): arch: stm32l4: Enable gpio interrupts correctly
Enable gpio interrupts correctly and don't
disable the already enabled interrupts.

Fixes #9630
commit d2e713f9 (2018-08-29): arch: stm32l4: Enable gpio interrupts correctly

Enable gpio interrupts correctly and don't
disable the already enabled interrupts.

Fixes #9630
closed 2018-08-29 by nashif