zephyrproject-rtos/zephyr #9651: Shell_module@mimxrt1050_evk runs failure on R1.13_RC1

hakehuang, 2018-08-27:
build Shell_module at samples/subsys/shell/shell_module with below command line
`cmake -DBOARD=mimxrt1050_evk  ..`
run on the mimxrt1050_evk  board, get below console output

> ======== Target Main Serial Content:   ========
start application at 0x2d4d !!!

***** HARD FAULT *****

  Fault escalation (see below)

***** USAGE FAULT *****

  Unaligned memory access

***** Hardware exception *****

Current thread ID = 0x20000344

Faulting instruction address = 0xd8a

Fatal fault in thread 0x20000344! Aborting.

***** HARD FAULT *****

  Fault escalation (see below)

***** USAGE FAULT *****

  Unaligned memory access

***** Hardware exception *****

Current thread ID = 0x20000344

Faulting instruction address = 0xd8a

Fatal fault in thread 0x20000344! Aborting.

***** HARD FAULT *****

  Fault escalation (see below)

***** USAGE FAULT *****

  Unaligned memory access

***** Hardware exception *****

Current thread ID = 0x20000344

Faulting instruction address = 0xd8a

Fatal fault in thread 0x20000344! Aborting.

***** HARD FAULT *****

  Fault escalation (see below)

***** USAGE FAULT *****

  Unaligned memory access

***** Hardware exception *****

Current thread ID = 0x20000344

Faulting instruction address = 0xd8a

Fatal fault in thread 0x20000344! Aborting.
[... 9773 lines cut ...]
Faulting instruction address = 0xd8a

Fatal fault in thread 0x20000344! Aborting.

***** HARD FAULT *****

  Fault escalation (see below)

***** USAGE FAULT *****

  Unaligned memory access

***** Hardware exception *****

Current thread ID = 0x20000344

Faulting instruction address = 0xd8a

Fatal fault in thread 0x20000344! Aborting.

***** HARD FAULT *****

  Fault escalation (see below)

*

hakehuang, 2018-08-27:
cc @MaureenHelm

MaureenHelm, 2018-08-27:
I'm not able to reproduce this. I got the following console output:
```
Zephyr Shell, Zephyr version: 1.13.0-rc1
Type 'help' for a list of available commands
shell> help
[Modules]
sample_module
kernel

[Commands]
version
noprompt

To select a module, enter 'select <module name>'.
shell> version
Zephyr version 1.13.0-rc1
shell> select sample_module
sample_module> help
help
ping                         
params                       print argc

Enter 'exit' to leave current module.
sample_module> ping
pong
sample_module> help params
Usage: params print argc
sample_module> params print argc
argc = 3
  argv[0] = params
  argv[1] = print
  argv[2] = argc
sample_module> 
```
Can you try doing a pristine build or POR-ing the board? I've seen the jlink connection occasionally get corrupted on this board.

PR #9865 (2018-09-11): arm: Clear primask during startup for all architecture variants
A bootloader may leave primask set, so clear it during startup when we
enable interrupts and switch to the main thread. Previously we only did
this for architecture variants which don't support basepri, but now we
do it for all architecture variants.

Fixes a failure on mimxrt1050_evk with the latency_measure test and
shell_module sample when using an nxp internal bootloader.

Fixes #9650 
Fixes #9651
closed 2018-09-11 by nashif