zephyrproject-rtos/zephyr #9656: tests/kernel/fp_sharing failing on sam_e70_xplained

spoorthik, 2018-08-27:
Execution log:

```
Floating point sharing tests started
===================================================================
FAIL - calculate_pi_low@89. Computed pi 3.141607, reference pi 3.141607
FAIL - load_store_low
===================================================================
===================================================================
PROJECT EXECUTION FAILED
```

Latest commit on which failure is observed: d8d5ec3f913e69b8b2d3bf46692da818567402d9

The issue is not reproducible always, the issue is seen in 3/10 runs when I ran on 2 different sam_e70_xplained targets.

Steps to reproduce:
```
cd tests/kernel/fp_sharing
mkdir build && cd build
cmake -DBOARD=sam_e70_xplained ..
make flash
```

nashif, 2018-08-29:
can't reproduce 

```
0/1 sam_e70_xplained          tests/kernel/fp_sharing/kernel.fp_sharing          building
MAKE: 'sanity_test_finished sam_e70_xplained/tests/kernel/fp_sharing/kernel.fp_sharing'
DEVICE: ***** Booting Zephyr OS v1.13.0-rc1-67-g521b256049 *****
DEVICE: Floating point sharing tests started
DEVICE: ===================================================================
DEVICE: Pi calculation OK after 50 (high) + 37 (low) tests (computed 3.141607)
DEVICE: Load and store OK after 100 (high) + 3130 (low) tests
DEVICE: Pi calculation OK after 150 (high) + 114 (low) tests (computed 3.141607)
DEVICE: Load and store OK after 200 (high) + 6260 (low) tests
DEVICE: Pi calculation OK after 250 (high) + 190 (low) tests (computed 3.141607)
DEVICE: Load and store OK after 300 (high) + 9390 (low) tests
DEVICE: Pi calculation OK after 350 (high) + 267 (low) tests (computed 3.141607)
DEVICE: Load and store OK after 400 (high) + 12520 (low) tests
DEVICE: Pi calculation OK after 450 (high) + 343 (low) tests (computed 3.141607)
DEVICE: Load and store OK after 500 (high) + 15650 (low) tests
DEVICE: PASS - load_store_high
DEVICE: ===================================================================
DEVICE: ===================================================================
DEVICE: PROJECT EXECUTION SUCCESSFUL
1/1 sam_e70_xplained          tests/kernel/fp_sharing/kernel.fp_sharing          PASSED
1 of 1 tests passed with 0 warnings in 42 seconds
```

spoorthik, 2018-08-30:
@nashif : I can still reproduce on the latest commit with TCF infrastructure.

Note: This adds 1000ms of boot delay.

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.13.0-rc1-113-g643c8ab (delayed boot 1000ms) *****
Floating point sharing tests started
===================================================================
FAIL - calculate_pi_low@89. Computed pi 3.141607, reference pi 3.141607
FAIL - load_store_low
===================================================================
===================================================================
PROJECT EXECUTION FAILED
```

nashif, 2018-08-31:
then you should reopen it.

nashif, 2018-08-31:
it is sporadic...

```
AKE: 'sanity_test_building sam_e70_xplained/tests/kernel/fp_sharing/kernel.fp_sharing'
0/1 sam_e70_xplained          tests/kernel/fp_sharing/kernel.fp_sharing          building
MAKE: 'sanity_test_finished sam_e70_xplained/tests/kernel/fp_sharing/kernel.fp_sharing'
DEVICE: Floating point sharing tests started
DEVICE: ===================================================================
DEVICE: Pi calculation OK after 50 (high) + 37 (low) tests (computed 3.141607)
DEVICE: Load and store OK after 100 (high) + 3130 (low) tests
DEVICE: Pi calculation OK after 150 (high) + 114 (low) tests (computed 3.141607)
DEVICE: Load and store OK after 200 (high) + 6260 (low) tests
DEVICE: Pi calculation OK after 250 (high) + 190 (low) tests (computed 3.141607)
DEVICE: Load and store OK after 300 (high) + 9390 (low) tests
DEVICE: Pi calculation OK after 350 (high) + 267 (low) tests (computed 3.141607)
DEVICE: Load and store OK after 400 (high) + 12520 (low) tests
DEVICE: Pi calculation OK after 450 (high) + 343 (low) tests (computed 3.141607)
DEVICE: Load and store OK after 500 (high) + 15650 (low) tests
DEVICE: PASS - load_store_high
DEVICE: ===================================================================
DEVICE: ===================================================================
DEVICE: PROJECT EXECUTION SUCCESSFUL
1/1 sam_e70_xplained          tests/kernel/fp_sharing/kernel.fp_sharing          PASSED
1 of 1 tests passed with 0 warnings in 41 seconds
```

aurel32, 2019-02-10:
This issue is quite difficult to trigger, I have been able to trigger it only once over 100 of times using old revision 643c8abea3b8826a9c52a35d49a12c216bdf425e. Therefore I went to look at the code protected by `CONFIG_FP_SHARING`:

https://github.com/zephyrproject-rtos/zephyr/blob/1d8a0bc1aad4412ec5954a1fc2f9fb5e8a9eef8b/arch/arm/core/swap_helper.S#L92-L95 and https://github.com/zephyrproject-rtos/zephyr/blob/1d8a0bc1aad4412ec5954a1fc2f9fb5e8a9eef8b/arch/arm/core/swap_helper.S#L180-L183

It looks odd to me that we only save s16-s31. Those are the one that must be preserved across subroutine calls, but from what I understand task switching can happen at any moment, so I think we need to save and restore **all** FP registers.

If that's actually the issue, I believe it also affects Cortex-M4F SoC, however the Pi computation is done using double, so it doesn't use FP registers for the computation, only to pass values between subroutines. We probably wants to add tests using float instead double.

ioannisg, 2019-03-20:
>It looks odd to me that we only save s16-s31. Those are the one that must be preserved across subroutine calls, but from what I understand task switching can happen at any moment, so I think we need to save and restore all FP registers.

@aurel32 , if i understand the specification correctly, the S0-S15 and FPSCR are stacked and unstacked automatically, during exception entry and return, respectively. So we do not need to save and restore these ones.

ioannisg, 2019-03-20:
I am running into troubles, trying to reproduce this one with current master:
- the test passes in nRF53840_pca10056 (Cortex-M4F), but takes around 15 minutes to execute; does this sound normal?
- the test cannot execute in sam_e70_xplained; the MCU resets after roughly 18 seconds; could it be an issue with the watchdog, @aurel32 ?

aurel32, 2019-03-20:
> @aurel32 , if i understand the specification correctly, the S0-S15 and FPSCR are stacked and unstacked automatically, during exception entry and return, respectively. So we do not need to save and [...]

For the exceptions I agree, but we also need to save and restore those registers when switching from one task to another.

> I am running into troubles, trying to reproduce this one with current master:
> 
>     * the test passes in nRF53840_pca10056 (Cortex-M4F), but takes around 15 minutes to execute; does this sound normal?

Have you changed the double into float? The M4F only has a single precision FPU, so the FP registers are not used, soft float is used instead.
 
>     * the test cannot execute in sam_e70_xplained; the MCU resets after roughly 18 seconds; could it be an issue with the watchdog, @aurel32 ?

This could be an issue with commit 2e01e86bdc70591c608caa9072de17226ede9e76. You can try to enable WDT_DISABLE_AT_BOOT to see if it is linked to the watchdog (it looks like that).

ioannisg, 2019-03-20:
> For the exceptions I agree, but we also need to save and restore those registers when switching from one task to another.

With tasks, you mean threads? If so, I agree, but this is already done since switching occurs in PendSV which is an exception, so the FP context is stacked in the stack frame of the switched-off thread, and restored from the stored stacked frame of the switched-in thread :)

ioannisg, 2019-03-20:
>This could be an issue with commit 2e01e86. You can try to enable WDT_DISABLE_AT_BOOT to see if it is linked to the watchdog (it looks like that).

@aurel32 will take a look on this, thanks.

ioannisg, 2019-03-21:
@aurel32 thanks for your suggestion; i disabled the Watchdog as you indicated, and now the test is able to execute and seems to pass.
I have remaining issues:
- the tests seems to fail immediately after the MCU reboots
- then it shows that is passes
- the MCU restarts after tests completion, so it keeps executing the test

This is the kind of output I am getting:

```
***** Booting Zephyr OS v1.14.0-rc1-1357-g64dba11f05a9 *****
Floating point sharing tests started
===================================================================
FAIL - calculate_pi_low@89. Computed pi %.6f, reference pi %.6f
FAIL - load_store_low
===================================================================
===================================================================
PROJECT EXECUTION FAILED
Pi calculation OK after 50 (high) + 0 (low) tests (computed %.6f)
Load and store OK after 100 (high) + 11619 (low) tests
Pi calculation OK after 150 (high) + 0 (low) tests (computed %.6f)
Load and store OK after 200 (high) + 11619 (low) tests
Pi calculation OK after 250 (high) + 0 (low) tests (computed %.6f)
Load and store OK after 300 (high) + 11619 (low) tests
Pi calculation OK after 350 (high) + 0 (low) tests (computed %.6f)
Load and store OK after 400 (high) + 11619 (low) tests
Pi calculation OK after 450 (high) + 0 (low) tests (computed %.6f)
Load and store OK after 500 (high) + 11619 (low) tests
PASS - load_store_high
===================================================================
===================================================================
PROJECT EXECUTION SUCCESSFUL
Pi calculation OK after 550 (high) + 0 (low) tests (computed %.6f)
Pi calculation OK after 650 (high) + 0 (low) tests (computed %.6f)
Pi calculation OK after 750 (high) + 0 (low) tests (computed %.6f)

```

agansari, 2019-03-22:
@ioannisg or @aurel32 can you please check if #14830 fixes this issue?

ioannisg, 2019-03-22:
> 
> 
> @ioannisg or @aurel32 can you please check if #14830 fixes this issue?

Yes, this fixes the initial failure. Thanks for the fix.

ioannisg, 2019-03-22:
I wouldn't close this issue until we clarify the issue with the Watchdog that needs to be disabled at boot, if we want the test to pass in SAM E70 XPLAINED. Or, at least, track the WDT issue somewhere else, as this is unrelated to FPU....

ioannisg, 2019-03-22:
@nashif @aurel32 the un-initialised FPCSR in the artificial stack frame, explains, IMO, the sporadic failures you had been observing

MaureenHelm, 2019-03-25:
Fixed by #14830

closed 2019-03-25 by MaureenHelm