zephyrproject-rtos/zephyr #9663: In many places, Zephyr uses signed type (s32_t) for elapsed time and delays, contradicting common sense, risking hitting C undefined behavior and/or MISRA C violations

pfalcon, 2018-08-27:
We have (among number of others):

~~~
void _nano_sys_clock_tick_announce(s32_t ticks);
~~~

Using s32_t here doesn't make sense, this function is "announce" to let know how many ticks elapsed since since previous announcement, which is always a non-negative value. Passing negative value here would make system time go back, and is not handled in any way by the function. 

~~~
extern int k_delayed_work_submit_to_queue(struct k_work_q *work_q,
                                          struct k_delayed_work *work,
                                          s32_t delay);
~~~

Likewise, delay can't really be negative. This function could assign some special meaning to negative delays (e.g. overdue delays, to save caller from detecting such), but it does not, the only special processing is for delay of 0:

~~~
        if (!delay) {
                /* Submit work if no ticks is 0 */
                k_work_submit_to_queue(work_q, &work->work);
        } else {
                /* Add timeout */
                _add_timeout(NULL, &work->timeout, NULL,
                                _TICK_ALIGN + _ms_to_ticks(delay));
        }
~~~

pfalcon, 2018-08-27:
@nashif, @andrewboie, @andyross: Would be be interesting to hear your opinion on this issue.

I think we should make one of 2 things:

1. Either document explicitly what happens if negative value is used (among choices: undefined behavior, treated as 0, error returned, treated as unsigned value).
2. Convert to u32_t.

My choice would be 2, because it allows to get more delay time out of 32-bit milliseconds value (~49 days instead of ~24 days). But that's obviously "API change" and requires making some assumptions on how overdue a timeout can be.

Making such an assumption shouldn't be a practical problem, but of course, some may see that there's a theoretical downside in the need of making such decision (vs accepting just maximum possible overdue of INT_MAX). Just the same as I see theoretical downside in limiting delays to 31 bits, while they can be easily 32 bits.

pfalcon, 2018-08-27:
How I got into that is that it causes artifacts in networking code, e.g.: https://github.com/zephyrproject-rtos/zephyr/pull/9659/commits/521409d8e540b30c620fdf2e032094ba885e6cb6#diff-11516a1156142dc5ebd6a6437176df96

cfriedt, 2021-07-20:
@pfalcon - is this still an issue?