zephyrproject-rtos/zephyr #9664: tests/kernel/threads/thread_apis/kernel.threads.user_mode crases on QEMU-x86

inakypg, 2018-08-27:
Previous subcases pass.

Toolchain: ZEPHYR_GCC_VARIANT=zephyr

Commit ID f72c4c52363920e225b4eb20cdbda0e620c1965e

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.13.0-rc1 (delayed boot 1000ms) *****
Running test suite threads_lifecycle
===================================================================
starting test - test_threads_spawn_params
PASS - test_threads_spawn_params
===================================================================
starting test - test_threads_spawn_priority
PASS - test_threads_spawn_priority
..
===================================================================
starting test - test_threads_spawn_delay
PASS - test_threads_spawn_delay
===================================================================
starting test - test_threads_spawn_forever
PASS - test_threads_spawn_forever
===================================================================
starting test - test_thread_start
PASS - test_thread_start
===================================================================
starting test - test_threads_suspend_resume_cooperative
PASS - test_threads_suspend_resume_cooperative
===================================================================
starting test - test_threads_suspend_resume_preemptible
PASS - test_threads_suspend_resume_preemptible
===================================================================
starting test - test_threads_priority_set
PASS - test_threads_priority_set
===================================================================
starting test - test_threads_abort_self
PASS - test_threads_abort_self
===================================================================
starting test - test_threads_abort_others
PASS - test_threads_abort_others
===================================================================
starting test - test_threads_abort_repeat
PASS - test_threads_abort_repeat
===================================================================
starting test - test_abort_handler
***** Stack Check Fail! *****
Current thread ID = 0x004180c0
eax: 0x00418420, ebx: 0x00408004, ecx: 0x00000000, edx: 0x00000000
esi: 0x00000000, edi: 0x00418420, ebp: 0x0040800c, esp: 0x00408000
eflags: 0x00000246 cs: 0x0008
call trace:
eip: 0x00002bad
     0x00002d73 (0x100)
     0x0000581f (0x418420)
     0x0000594a (0x418420)
     0x00005996 (0x418420)
     0x00001c23 (0x40)
     0x00002b1b (0x0)
Fatal fault in thread 0x004180c0! Aborting.
Child thread's abort handler called
Calling abort of child from parent
Child thread's abort handler called
PASS - test_abort_handler
===================================================================
starting test - test_delayed_thread_abort
PASS - test_delayed_thread_abort
```

inakypg, 2018-08-27:
Note this was executed with a CONFIG_BOOT_DELAY of 1000, which might be helping trigger the issue

nashif, 2018-08-27:
just a stack size issue, fix coming..

PR #9672 (2018-08-27): tests: thread_api: increase stack for test
Add more stack for this test, it was failing and hidden by sanitycheck
(which needs to be fixed somewhere else).

Fixes #9664
commit d2b4d8f0 (2018-08-27): tests: thread_api: increase stack for test

Add more stack for this test, it was failing and hidden by sanitycheck
(which needs to be fixed somewhere else).

Fixes #9664
closed 2018-08-27 by nashif