zephyrproject-rtos/zephyr #9665: tests/drivers/watchdog/wdt_basic_api crashes on Quark D2k / SE and ESP32

inakypg, 2018-08-27:
Toolchain: zephyr / espressif

Commit ID f72c4c52363920e225b4eb20cdbda0e620c1965e

On Quark SE / x86

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.13.0-rc1 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
***** CPU exception 7
Current thread ID = 0xa80080a0
eax: 0xa8008101, ebx: 0xa8008104, ecx: 0x00000000, edx: 0xa800702c
esi: 0xafdc7887, edi: 0x40030505, ebp: 0xa800753c, esp: 0xa8007534
eflags: 0x00010282 cs: 0x0008
eip: 0x40000024
Fatal fault in thread 0xa80080a0! Aborting.
```

On Quark D2000:

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.13.0-rc1 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
***** CPU exception 6
Current thread ID = 0x00280cc0
eax: 0x00280d18, ebx: 0x00280d18, ecx: 0x00000000, edx: 0x0028002c
esi: 0x00000000, edi: 0x001804ef, ebp: 0x0028055c, esp: 0x00280554
eflags: 0x00010296 cs: 0x0008
eip: 0x00000000
Fatal fault in thread 0x00280cc0! Aborting.
```

On ESP32:

```
***** Booting Zephyr OS 1.13.0-rc1 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
 ** FATAL EXCEPTION
 ** CPU 0 EXCCAUSE 20 PS 0x00060120 PC 0x00000000 VADDR 0x00000000
 **  A0 0x800809bb  SP 0x3ffb1e60  A2 0x3ffb0000  A3 0x3ffb04ec
 **  A4 0x3ffb03b6  A5 0x3ffb29f0  A6 0x3ffb021c  A7 0x3ffb17d0
 **  A8 0x80080948  A9 0x3ffb1df0 A10 0x3ffb0000 A11 0x3ffb1598
 ** A12 0x3ffb1e40 A13 0x00000004 A14 0x00060220 A15 0x3ffb17d0
 ** LBEG 0x00000000 LEND 0x00000000 LCOUNT 0x00000000
 ** SAR 0x00000000
Current thread ID = 0x3ffb021c
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
```

nashif, 2018-08-27:
duplicate of #8097

closed 2018-08-27 by nashif

inakypg, 2018-08-27:
Ooops, sorry -- missed double checking this one.