zephyrproject-rtos/zephyr #9666: tests/benchmarks/timing_info/testcase.yaml#benchmark.timing.default_kernel crashes on Arduino 101 / ARC

inakypg, 2018-08-27:
Toolchain: ZEPHYR_GCC_VARIANT=zephyr

Commit ID f72c4c52363920e225b4eb20cdbda0e620c1965e

```
***** delaying boot 1000ms (per build configuration) *****
Exception vector: 0xd, cause code: 0x0, parameter 0x0
Address 0xb59bee47
Current thread ID = 0xa8000474
Faulting instruction address = 0x40036bbc
Fatal fault in essential thread! Spinning...
```

Note this was built with CONFIG_BOOT_DELAY of 1000.

AdithyaBaglody, 2018-09-03:
@inakypg how are you running this? 
Because i flashed the code benchmarking on ARC and a hello world on x86. It works.

inakypg, 2018-09-04:
default config + CONFIG_BOOT_DELAY=1000

AdithyaBaglody, 2018-09-10:
inaky, try running this with a simple hello world project on x86 (with config_arc_init)
and run the timing_info on the arc. It works fine.
This is happening because some random application is running on x86. Which might interfere with the code execution on the ARC.

closed 2018-09-26 by nashif