zephyrproject-rtos/zephyr #9670: Bluetooth: Mesh: Persistent Storage: Node address not restored

MrPointer, 2018-08-27:
Hi all. 
In the last couple of days my Team and I are attempting to use the BT-Mesh-Settings API, with our current implementation being based on self-provisioning and self-configuring (We have our reasons). After a bit of grappling, we found out the updated [mesh_demo](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/bluetooth/mesh_demo) (We've had an offline copy of v1.12) which does exactly what we've been looking after. 
However, the problem is it doesn't work for us as expected.

The Node's address, or more specifically, the Primary Element's address (As we don't have any more for now) is not restored upon a power-cycle, leading to error `-49:EADDRNOTAVAIL` when publishing data using a publication context/model.
Since we're self-provisioning, we could "overcome" this issue by manually calling the `bt_mesh_comp_provision` function, which from our understanding should assign the address - But it's an ugly solution and wouldn't work for prod-ready provisioning.

Now the question is - Is the address not restored on purpose? It seems rather foolish to restore settings on boot after a system reset of some kind just to wait for the node to be re-provisioned, isn't it?
If it's not, I'm afraid this is some kind of bug, or perhaps, we're using it wrong?

jhedberg, 2018-08-27:
Storing and restoring the range of assigned addresses should work fine, and does work fine using the existing sample apps in the upstream tree. The relevant code is in `subsys/bluetooth/host/mesh/settings.c` where `net_set()` is responsible for restoring and `store_pending_net()` for storing. These deal with the device key and the primary element address (from which the remaining elements, if any, are derived).

I'm a bit confused by what you're doing - you say you're self-provisioning. What exactly do you mean by that? What APIs do you call and in what order?

jhedberg, 2018-08-28:
@MrPointer what value does your `bt_mesh_provision()` call return?

jhedberg, 2018-08-28:
@MrPointer also, you earlier mentioned `bt_mesh_comp_provision()` - did you mean `bt_mesh_provision()` instead? If you're calling the stack-private `bt_mesh_comp_provision()` function that could explain some of the behavior you're seeing.

MrPointer, 2018-08-28:
@jhedberg No no, I don't call the `bt_mesh_comp_provision` function, don't worry :laughing: 
I mentioned it as a possible solution we've thought of, in case there's indeed a bug.

About the `bt_mesh_provision` function - On the first run after flashing the board i.e before any settings are persisted, the function exits successfully with error code **0**. 
On the 2nd run, after power-cycling i.e settings have been already persisted and loaded, it returns with error code **EALREADY**, which is exactly what we expect.

jhedberg, 2018-08-28:
@MrPointer are you sure your primary element's address is unassigned? There is no place in the mesh code that would return EADDRNOTAVAIL for that situation. That said, the publishing functions *will* return EADDRNOTAVAIL if you haven't configured a publishing address, or if there's no app key to use for publishing.

MrPointer, 2018-08-28:
@jhedberg Actually I'm not sure about that, however, about the other you mentioned - It might possibly be the problem here, I'll need to do some checks.
Theoritically shouldn't the publication address be stored and restored? 
Because I'm preety sure the App Key is persisted... I'm not wrong am I?

jhedberg, 2018-08-28:
@MrPointer yes, both App Keys and Model Publication should be restored if previously set through the configuration model. Is that how you're configuring it? E.g. manually assigning an address to your publication context will not work since then the stack can't know that it needs to store something to flash.

MrPointer, 2018-08-28:
@jhedberg Hmmm... That seems to be it - We're manually assigning the publication address, as part of the self configuration. Isn't there a way to use publication contexts when self-configuring?

jhedberg, 2018-08-28:
@MrPointer yes, by using the configuration client API. I.e. `bt_mesh_cfg_mod_pub_set()` from `include/bluetooth/mesh/cfg_cli.h`. The mesh_demo app already uses several configuration client APIs, so I'm surprised you didn't use it for model publication as well.

MrPointer, 2018-08-29:
@jhedberg Sorry for the delay, have been busy with something else.
I think we misunderstood each other - When you said "manually", I thought you meant self-configuring.
That being said, we **are** using the `bt_mesh_cfg_mod_pub_set` function to configure the pub context.

Also, I've rerun our tests and the Node's address is indeed restored as expected. Publication address however, is not.

jhedberg, 2018-08-29:
I did some testing with samples/bluetooth/mesh_demo, and I am not able to reproduce this issue - the publication address gets correctly restored after a power cycle. I did however have to increase some buffer counts for samples/bluetooth/mesh_demo to make the Model Pub Set & Model Pub Status messages get sent & processed properly:

```
CONFIG_BT_MESH_TX_SEG_MSG_COUNT=3
CONFIG_BT_MESH_ADV_BUF_COUNT=14
```

Also note that the processing of the configuration messages happens through a queue, so won't see it take effect immediately when the `bt_mesh_cfg_mod_pub_set()` call returns.

jhedberg, 2018-08-29:
One more tip: if you haven't done so already, please enable `CONFIG_BT_DEBUG_LOG=y` so that you can see any error & warning logs from the stack.

MrPointer, 2018-08-29:
@jhedberg Yes I know that thing about the queue, although it always seems that none of its' element is parsed until the `bt_ready` functions exits, which is somewhat strange (or not if it's the same thread processing them, this is internal and I hadn't looked at it).
I'm not near the code right now so I can't check the `prj.conf`, however, I surely remember setting the `CONFIG_BT_MESH_TX_SEG_MSG_COUNT` to 3, so just the `CONFIG_BT_MESH_ADV_BUF_COUNT` needs to be validated. Will edit once I'll check.

Lastly, regarding the logging tip - I'm aware of that as well but its' quite inconvenient to do when testing on a completely offline machine, not to mention porting logs outside 😅 
I'll try to attach meaningful logs nevertheless.

jhedberg, 2018-08-29:
@MrPointer you can do the configuration client calls blockingly, by passing a non-NULL status parameter (usually the last parameter). That means "I'm interested in the final status, only return once you have received the response". The challenge with this is that the local mesh network interface is processed through the system workqueue, which happens to be the same context that the bt_ready callback gets called in. So you can't do any blocking configuration client calls directed at the local node from the bt_ready callback, since it will just result in a deadlock and a timeout when waiting for the response (the default wait is 2 seconds).

MrPointer, 2018-08-29:
@jhedberg Thanks for the explanation, it makes sense.
On a side note, could this be documented? I haven't seen it anywhere in the documentation nor the API, along with all other configuration methods, which is quite oppressive...

jhedberg, 2018-08-29:
@MrPointer any contributions to the configuration client API documentation are welcome ;)

There are some reasons I haven't gotten around to it yet:

 - It's huge, and for the most part the API maps 1:1 to what's already documented in the Bluetooth Mesh Profile specification (Section 4).
 - It's not really useful for anything else than talking to the local node, which is something I consider a hack/demo use case (hence documentation would be good, but it's not as important as the other APIs).

The configuration client API will for the most part work for talking to other nodes, but it fails at the stage of selecting the right Device Key. Currently the stack only knows about its own Device Key that was assigned to it by a provisioner. Knowing other nodes' Device Keys implies being a provisioner ourselves (or having been delegated the task of configuring other nodes by a provisioner) which is something that stack has no support for (yet - I welcome it if someone wants to pick up this task).

MrPointer, 2018-08-29:
@jhedberg This is very reasonable of course, after all my usage is not very standard, and wouldn't be when project will be ready for production.
Had I had the time, I would happily documented it. Maybe someday I will 😉 

Back to topic's issue - It seems I won't be able to test anything today and my peers are also away from work. Hopefully tomorrow we'll return with some answers 😃 
@RonAmihai Please review [this](https://github.com/zephyrproject-rtos/zephyr/issues/9670#issuecomment-416892218) when you have time and return to us with an update.

RonAmihai, 2018-08-29:
Our -EADDRNOTAVAIL actually appears inside the bt_mesh_model_publish function, after our custom publication callback (the 'update' field of the publication context structure) is called. 

Inside bt_mesh_model_publish, there is a call to bt_mesh_app_key_find, which returns NULL and therefore bt_mesh_model_publish returns -EADDRNOTAVAIL

RonAmihai, 2018-08-29:
@jhedberg Just need to mention once again: 
Start from the first power-cycle we are not calling the bt_mesh_cfg_mod_pub_set, since there are stored BT settings therefore configuration shouldn't be needed. 

The above error appears only after the first power cycle. 
The first boot, which calls bt_mesh_cfg_mod_pub_set, works perfectly and the publication callback publishes without issues in this case.

jhedberg, 2018-08-29:
So now we've gone from the local unicast address missing, to the publication (destination) address missing, to there not being an app key? :)

Could someone update the topic of this issue accordingly? (I already did it once)

jhedberg, 2018-09-02:
I'm unable to reproduce the App Key issue. I added some printk calls, without any other modifications, to the upstream mesh_demo, and it shows that AppKey Index 0 is correctly restored to the internal App Key list as well as the vendor model binding.

jhedberg, 2018-09-02:
And for the record, the printk calls I added look like this: http://ix.io/1lNX

MrPointer, 2018-09-02:
@jhedberg We've got some updates which might give more insight into the issue.

We've noticed an internal call to `bt_mesh_app_key_find` always returns **NULL** after a power-cycle, i.e On the 2nd run/boot.
When sending a message through a publication context, the error -**EADDRNOTAVAIL** is returned as a result of an internal call to `bt_mesh_app_key_find`.
When sending a manual access-layer message (`bt_mesh_model_send`), the error -**EINVAL (-22)** is returned as a result of an internal call to `bt_mesh_app_key_find`.

We're attempting to attach some more logging output now to give even more insight on what's happening.

RonAmihai, 2018-09-02:
[Log1 (First Boot).txt](https://github.com/zephyrproject-rtos/zephyr/files/2343472/Log1.First.Boot.txt)
[Log2 (After Power Cycle).txt](https://github.com/zephyrproject-rtos/zephyr/files/2343475/Log2.After.Power.Cycle.txt)
[prj.conf.txt](https://github.com/zephyrproject-rtos/zephyr/files/2343476/prj.conf.txt)

@jhedberg I've attached few logs, alongside our proj conf file.
 
The first log represents the first boot of the device, without any previous stored BT settings. It continuously publishing data (5s intervals). I've captured only the first publish events. 

The second log represents boots after the first power cycle - with previous stored BT settings. The device just fails on the first publishing and hangs in this situation.

jhedberg, 2018-09-02:
Actually, I'm not sure it's valid to rely on the ordering of settings keys, so I've created a pull request to remove this check: #9760

If it fixes the issue for you please comment in the PR so that we can get it merged.

PR #9760 (2018-09-03): Bluetooth: Mesh: Fix checking for subnet when recovering App Key
The ordering of items in flash is not guaranteed, so it's possible we
get an App Key before the corresponding Net Key. Remove the check for
a Net Key, since the storing code should never store an App Key if
there is no corresponding Net Key.

Fixes #9670
commit 697919c5 (2018-09-03): Bluetooth: Mesh: Fix checking for subnet when recovering App Key

The ordering of items in flash is not guaranteed, so it's possible we
get an App Key before the corresponding Net Key. Remove the check for
a Net Key, since the storing code should never store an App Key if
there is no corresponding Net Key.

Fixes #9670
closed 2018-09-03 by nashif