zephyrproject-rtos/zephyr #9671: Zephyr with WNC-M14A2A not compiling

derchrismakaio, 2018-08-27:
Hi, it seems like there are missing defines when using WNC-M14A2A:
zephyr/include/generated/generated_dts_board.h:185:50: error: 'NORDIC_NRF_UARTE_40028000_WNCM14A2A_MDM_KEEP_AWAKE_GPIOS_PIN' undeclared here (not in a function)

(just an example)

mike-scott, 2018-08-27:
@derchrismakaio Which board were you testing this with?

derchrismakaio, 2018-08-27:
I'm porting it to nrf52_pca10040. I think it should work with CONFIG_UART_CONSOLE=n.

I've changed uart0 in nrf52_pca10040.dts:
`&uart0 {
	status = "ok";
	compatible = "nordic,nrf-uart";
	current-speed = <115200>;

	#ifdef CONFIG_MODEM_WNCM14A2A
	wncm14a2a {
		compatible = "wnc,m14a2a";
		label = "wncm14a2a";
		mdm-boot-mode-sel-gpios = <&gpio0 2 0>;
		mdm-power-gpios = <&gpio0 3 0>;
		mdm-keep-awake-gpios = <&gpio0 7 0>;
		mdm-reset-gpios = <&gpio0 10 0>;
		mdm-shld-trans-ena-gpios = <&gpio0 11 0>;
		status = "ok";
	};
	#endif
};`

And, of course,  I've enabled the modem via menu

Added to prj.conf of lwm2m_client sample:
`CONFIG_CONSOLE_SHELL=n
CONFIG_UART_CONSOLE=n
CONFIG_MODEM_RECEIVER_MAX_CONTEXTS=1
CONFIG_MODEM_WNCM14A2A=y`

mike-scott, 2018-08-27:
@derchrismakaio I would follow the changes here: https://github.com/zephyrproject-rtos/zephyr/commit/65af15bcc2e9fb1936505ce95f543f569b51b44a#diff-115b4ae91af8f499ac18e8b572738d7a  but apply them to boards/arm/nrf52_pca10040.

Specifically, the error you're seeing is related to the dts.fixup file changes in the above patch.

mike-scott, 2018-08-27:
Ah, nevermind.. I see the issue.
You'll need to also change this in the prj.conf:
CONFIG_UART_0_NRF_UART=n
CONFIG_UART_0_NRF_UARTE=y

derchrismakaio, 2018-08-27:
yepp, funny, I did that ;) but now I get 

Error: Aborting due to non-whitelisted Kconfig warning 'warning: the choice symbol UART_0_NRF_UARTE (defined at drivers/serial/Kconfig.nrfx:34) is selected by the following symbols, which has no effect: 
 - MODEM_WNCM14A2A (defined at drivers/modem/Kconfig:42)'

derchrismakaio, 2018-08-27:
whooops, sorry, that was because of a change in modem's Kconfig:

from
select UART_1_NRF_UARTE if SOC_NRF52840
select UART_1_NRF_FLOW_CONTROL if SOC_NRF52840

to 

select UART_1_NRF_UARTE if SOC_NRF52840
select UART_0_NRF_UARTE if SOC_NRF52832
select UART_1_NRF_FLOW_CONTROL if SOC_NRF52840
select UART_0_NRF_FLOW_CONTROL if SOC_NRF52832

Apologies, these are my first 4 hours with Zephyr ;)

derchrismakaio, 2018-08-27:
still, it fails to compile because of the undefined "macros", starting with:
zephyr/include/generated/generated_dts_board.h:199:54: error: 'NORDIC_NRF_UARTE_40028000_WNCM14A2A_MDM_BOOT_MODE_SEL_GPIOS_CONTROLLER' undeclared here (not in a function)
 #define CONFIG_WNCM14A2A_GPIO_MDM_BOOT_MODE_SEL_NAME NORDIC_NRF_UARTE_40028000_WNCM14A2A_MDM_BOOT_MODE_SEL_GPIOS_CONTROLLER

prj.conf lines added:

CONFIG_CONSOLE_SHELL=y
CONFIG_UART_CONSOLE=y
CONFIG_MODEM_RECEIVER_MAX_CONTEXTS=1
CONFIG_MODEM_WNCM14A2A=y
CONFIG_UART_0_NRF_UART=n
CONFIG_UART_0_NRF_UARTE=y

going to bed now, phew, maybe it's easier to program it to a nrf52840 DK tomorrow in office

mike-scott, 2018-08-28:
@derchrismakaio I'll try your setup for nrf52_pca10040 in a bit, and see what's going on.

derchrismakaio, 2018-08-28:
@mike-scott tried to flash a pca10056 but I have no idea about the pin mapping - how do I find out what pin 46 on the board is? I've looked in Nordic's HAL but they're just saying to use the pin map macro, which doesn't help in this case, obviously :)

mike-scott, 2018-08-28:
@derchrismakaio Here's the pin mappings for nRF52840:
[P1.15/P0.47] MDM_UART2_CTS
[P1.14/P0.46] MDM_UART2_TX
[P1.13/P0.45] MDM_UART2_RX
[P1.12/P0.44] MDM_UART2_RTS
[P1.11/P0.43] SHIELD_3V3_1V8_SIG_TRANS_EN
[P1.10/P0.42] MDM_RESET
[P1.07/P0.39] MDM_WAKEUP_EN
[P1.03/P0.35] MDM_POWER_ON
[P1.02/P0.34] MDM_BOOT_MODE_SEL (on powerup, 0 = boot mode, 1 = normal boot)

You'll note that there are both P0 and P1 pin #'s as technically the pins are the P1 GPIO block, but the way the GPIO driver is setup, you can access them via P0 by using #'s higher than pin 31.

mike-scott, 2018-08-29:
@derchrismakaio I was able to get the nrf52_pca10040 working with JLinkRTTClient and this diff:
https://hastebin.com/ijunafeqex.diff

Here's a boot log: https://hastebin.com/oquleruzuf.log

NOTE: I found that we have a bug when using uart_fifo_fill() on the nRF52.  I'll need to debug that and see what I'm doing wrong there.  It works on the FRDM_K64F.

mike-scott, 2018-08-29:
Oh and I applied a local patch that I have yet to submit to the master branch:
https://hastebin.com/ematojiqet.patch

mike-scott, 2018-08-29:
For reference here are the pin mappings for nRF52832:
[P0.25] MDM_UART2_CTS
[P0.24] MDM_UART2_TX
[P0.23] MDM_UART2_RX
[P0.22] MDM_UART2_RTS
[P0.20] SHIELD_3V3_1V8_SIG_TRANS_EN
[P0.19] MDM_RESET
[P0.17] MDM_WAKEUP_IN
[P0.13] MDM_POWER_ON
[P0.12] MDM_BOOT_MODE_SEL

derchrismakaio, 2018-08-29:
Thanks @mike-scott, I‘ve got it up and running yesterday. I‘m trying to port some Quectel modems and it turned out that the biggest part of the problem was the lack of FC for my breakout.
What finally fixed it was to concat the newline and the AT command to be send at once.
If you want I can post the change here.
Also, are you fine with me copying your code for now?
Plus, the Quectels also work with mostly the same AT commands (except the % commands), so may I suggest to add another abstraction layer for AT command based modems and configure it by moving the modem-specific commands to a, for example, m95.h?

mike-scott, 2018-08-29:
@derchrismakaio I'm fine with the code copy.  Please submit a PR if you want and we can review code there.

Also feel free to submit a PR with the AT command abstraction layer.

Question: Do we feel this issue is still valid or should we close?

mike-scott, 2018-08-29:
Closing this issue as support for the modem using nrf52_pca10040 was never introduced (missing defines is normal in this case).

closed 2018-08-29 by mike-scott

derchrismakaio, 2018-08-29:
Is there a discussion group anywhere?
Some basic questions on the side if you don’t mind. Answers would help me understand my next steps for a clean integration better:
- say I want some app specific logic like a callback for an incoming call - where to place it best? Can I access the specific device (not the „interfaced“ object) from main code?
- is it easy to integrate bus selection for a driver in menuconfig? 
- is the hardcoding of default values in the board config the right way? Or is it easy to create the menu to select the specific pins for each device (in this case: the modem) and have ninja check and warn if same pins are assigned to different devices? I‘m just asking because, as you know, the nRF52840 can use every GPIO for each bus

mike-scott, 2018-08-29:
@derchrismakaio You can ask more general "development" style questions at devel@lists.zephyrproject.org

> say I want some app specific logic like a callback for an incoming call - where to place it best? Can I access the specific device (not the „interfaced“ object) from main code?

Depends on the kind of incoming call.  I might need more specifics or an example here.

> is it easy to integrate bus selection for a driver in menuconfig?

I do that here: https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/modem/Kconfig#L42 but this may change once the "shields" definitions are put into Zephyr.

> is the hardcoding of default values in the board config the right way? Or is it easy to create the menu to select the specific pins for each device (in this case: the modem) and have ninja check and [...]

Again the upcoming shield support may make this easier.  See: https://github.com/zephyrproject-rtos/zephyr/pull/6837  However, I may be misunderstanding your question and need more specifics.  I know that ninja won't have enough information to generated a warning if pins cross devices.

Also, I see "RFC" tagged PRs put up where a developer is looking to discuss a certain implementation as well.