zephyrproject-rtos/zephyr #9677: tests:cmsis_rtos_v1: test_mutex crashes with bus fault on sam_e70_xplained

spoorthik, 2018-08-28:
Execution log:
```
***** Booting Zephyr OS v1.13.0-rc1-54-g3b80998 (delayed boot 1000ms) *****
Running test suite test_cmsis_apis
===================================================================
starting test - test_kernel_start
PASS - test_kernel_start
===================================================================
starting test - test_kernel_systick
PASS - test_kernel_systick
===================================================================
starting test - test_thread_apis
PASS - test_thread_apis
===================================================================
starting test - test_thread_prio
PASS - test_thread_prio
===================================================================
starting test - test_thread_instances
PASS - test_thread_instances
===================================================================
starting test - test_timer
oneshot_callback (Timer 1) = 1
periodic_callback (Timer 2) = 1
periodic_callback (Timer 2) = 2
periodic_callback (Timer 2) = 3
periodic_callback (Timer 2) = 4
periodic_callback (Timer 2) = 5
periodic_callback (Timer 2) = 6
PASS - test_timer
===================================================================
starting test - test_mutex
***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0x316000c8
***** Hardware exception *****
Current thread ID = 0x20406450
Faulting instruction address = 0x4051f4
Fatal fault in thread 0x20406450! Aborting.
```
Latest commit on which issue is seen: 3b80998ff2204c9edecdae88a4849a6b26e7d06a

Platform: sam_e70_xplained

spoorthik, 2018-08-28:
This issue is possibly due to #9680

nashif, 2018-08-29:
fixed

closed 2018-08-29 by nashif