zephyrproject-rtos/zephyr #9689: Multiple tests are failing on sam_e70_xplained once the cache is enabled

spoorthik, 2018-08-28:
Due to commit c090776 multiple tests are failing on sam_e70_xplained.

spoorthik, 2018-08-28:
Refer: https://github.com/zephyrproject-rtos/zephyr/issues/9684, https://github.com/zephyrproject-rtos/zephyr/issues/9683, https://github.com/zephyrproject-rtos/zephyr/issues/9682, https://github.com/zephyrproject-rtos/zephyr/issues/9680, https://github.com/zephyrproject-rtos/zephyr/issues/9677 for execution logs.

tgorochowik, 2018-08-28:
That probably won't be very helpful, but if the original MPU/cache patches from #8185 are applied to the  `zephyr-v1.12.0` tag - the tests pass (and the cache works) (I remembered it to work and I just verified that).

So probably one of the recent MPU changes is causing all of this, not enabling the caches itself (but of course enabling them does trigger this bug).

nashif, 2018-08-28:
merged now

closed 2018-08-28 by nashif

aurel32, 2018-11-02:
I have tried to re-enable the caches on the current git code, and I am not able to reproduce all those issues. @spoorthik, can you try to re-apply the patch enabling the caches to check it is also fixed on your side?

aurel32, 2018-11-02:
I forgot to say I **do** reproduce the issue with the commit just before the revert.

aurel32, 2018-11-03:
@spoorthik, thanks for the test. I ran exactly the same thing, but I don't have the issue. I believe you are still interested to the cache enabled by default on the SAM E70 platform. I have therefore just opened the issue #11069 to discuss that, that is probably better than here.