zephyrproject-rtos/zephyr #9708: tests/kernel/tickless/tickless_concept fails on nRF5x with 1000msec boot delay

pswarnak, 2018-08-29:
The test **tests/kernel/tickless/tickless_concept** results in Assertion failure in test_tickless_slice subtestcase, when boot delay of 1000msec is added.

**Arch:** arm
**Board:** nrf51/52 - nrf51_pca10028:arm, nrf52_pca10040:arm, nrf52840_pca10056:arm
**Zephyr commit:** 6a01c69f9b94009e924ab56edf3e65dca97019bb
**Toolchain:**  ZEPHYR_TOOLCHAIN_VARIANT=zephyr
**Additional Info:** Add CONFIG_BOOT_DELAY=1000 in tests/kernel/tickless/tickless_concept/prj.conf

**Error Console Log:**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.13.0-rc1-69-g6a01c69 (delayed boot 1000ms) *****
Running test suite tickless_concept
===================================================================
starting test - test_tickless_sysclock
time 1027, 1237
time 1247, 1446
PASS - test_tickless_sysclock
===================================================================
starting test - test_tickless_slice
elapsed slice 110, expected: <99, 109>

    Assertion failed at /home/pswarnak/workspace/1.13_execution/zephyr/tests/kernel/tickless/tickless_concept/src/main.c:59: thread_tslice: t <= SLICE_SIZE_LIMIT is false

elapsed slice 10, expected: <99, 109>

    Assertion failed at /home/pswarnak/workspace/1.13_execution/zephyr/tests/kernel/tickless/tickless_concept/src/main.c:57: thread_tslice: t >= SLICE_SIZE is false

elapsed slice 20, expected: <99, 109>

    Assertion failed at /home/pswarnak/workspace/1.13_execution/zephyr/tests/kernel/tickless/tickless_concept/src/main.c:57: thread_tslice: t >= SLICE_SIZE is false

elapsed slice 20, expected: <99, 109>

    Assertion failed at /home/pswarnak/workspace/1.13_execution/zephyr/tests/kernel/tickless/tickless_concept/src/main.c:57: thread_tslice: t >= SLICE_SIZE is false

FAIL - test_tickless_slice
===================================================================
===================================================================
RunID: :r7eb
PROJECT EXECUTION FAILED
```

**Steps to reproduce:**
```
cd zephyr/tests/kernel/tickless/tickless_concept
rm -rf build && mkdir build && cd build
cmake -D BOARD=nrf51_pca10028 ../
make BOARD=nrf51_pca10028 flash

Check the console log.
```

pswarnak, 2018-08-29:
Hi @pizi-nordic,
Can you please check the failure, #9137 still does not resolve the failure, when boot delay of 1000msec is added.

pswarnak, 2018-08-29:
Just figured out that **tests/kernel/sched/schedule_api/testcase.yaml#kernel.sched.slice_reset** also fails by adding 1000msec boot delay on nRF5x boards. Please find console output below:
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.13.0-rc1 (delayed boot 1000ms) *****
Running test suite threads_scheduling
===================================================================
starting test - test_priority_cooperative
PASS - test_priority_cooperative
===================================================================
starting test - test_priority_preemptible
PASS - test_priority_preemptible
===================================================================
starting test - test_yield_cooperative
PASS - test_yield_cooperative
===================================================================
starting test - test_sleep_cooperative
PASS - test_sleep_cooperative
===================================================================
starting test - test_sleep_wakeup_preemptible
PASS - test_sleep_wakeup_preemptible
===================================================================
starting test - test_pending_thread_wakeup
PASS - test_pending_thread_wakeup
===================================================================
starting test - test_time_slicing_preemptible
PASS - test_time_slicing_preemptible
===================================================================
starting test - test_time_slicing_disable_preemptible
PASS - test_time_slicing_disable_preemptible
===================================================================
starting test - test_lock_preemptible
PASS - test_lock_preemptible
===================================================================
starting test - test_unlock_preemptible
PASS - test_unlock_preemptible
===================================================================
starting test - test_sched_is_preempt_thread
PASS - test_sched_is_preempt_thread
===================================================================
starting test - test_slice_reset

    Assertion failed at /home/jenkins/workspace/zephyr-master-tcf-v0.11-branch/LABEL/verify/SHARD/2-3/ZEPHYR_GCC_VARIANT/zephyr/zephyr.git/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:46: thread_tslice: t <= expected_slice_max is false

    Assertion failed at /home/jenkins/workspace/zephyr-master-tcf-v0.11-branch/LABEL/verify/SHARD/2-3/ZEPHYR_GCC_VARIANT/zephyr/zephyr.git/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:45: thread_tslice: t >= expected_slice_min is false

    Assertion failed at /home/jenkins/workspace/zephyr-master-tcf-v0.11-branch/LABEL/verify/SHARD/2-3/ZEPHYR_GCC_VARIANT/zephyr/zephyr.git/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:45: thread_tslice: t >= expected_slice_min is false

FAIL - test_slice_reset
===================================================================
starting test - test_slice_scheduling
PASS - test_slice_scheduling
===================================================================
starting test - test_priority_scheduling
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
PASS - test_priority_scheduling
===================================================================
starting test - test_wakeup_expired_timer_thread
PASS - test_wakeup_expired_timer_thread
===================================================================
===================================================================
RunID: ci-180828-1458-2110:mp9z
PROJECT EXECUTION FAILED
```

andyross, 2018-11-15:
Should be fixed as of the timer rework.   I just retried this on a nrf52840_pca10056, with and without BOOT_DELAY=1000 and it was reliable across 20 tries each.

carlescufi, 2018-12-05:
@pizi-nordic can you please re-run and close if fixed?

pizi-nordic, 2018-12-07:
I have tun 100 iterations without delay and 100 iterations with delay. There was no even single failure. Closing.

closed 2018-12-07 by pizi-nordic