zephyrproject-rtos/zephyr #9711: RFC: Zephyr should provide a unique id interface

therealprof, 2018-08-29:
Many (most?) MCUs provide some unique serial or ID factory programmed into the chip which could (and often should) be used to derive constant serial numbers, device identifiers, hostnames, MAC addresses etc. from.

At the moment applications and the kernel chose various details internally (e.g. hostname == board name, MAC address hardcoded or random, serial number hardcoded or application specifically read from non-volatile storage) which is neither portable nor comfortable nor does it allow the same application to run on different devices and represent themselves as such out-of-the-box and hence should be addressed.

There should be a driver interface which allows writing drivers which can obtain the device specific serialisation information and present it to the kernel. Examples for such drivers would be:
* MCU specific unique ID registers
* Dedicated serial number and security chips via I/O
* Vendor generated serialisation information in PROM or Flash
* Randomly generated
* Hardcoded via config
* Bootloader passthrough

There should also be an API in the kernel which allows  to retrieve that information in various useful formats, e.g.:
* binary bytes
* hexadecimal string representation
* UUID
* 32/64/128 bit values via integral types

And of course other interfaces and sample application should be adapted to use the unique id interface instead of guessing/hardcoding values or introducing their own way of configuration.

otavio, 2018-08-30:
We are doing it too, for updatehub support. Having a common way to do it does make a lot of sense.

therealprof, 2019-03-02:
CC #12372

aurel32, 2019-03-06:
Now that the hwinfo support has been merged, I guess this can be closed, correct?

therealprof, 2019-03-06:
@aurel32 Well, it certainly would help if there was a bit more API around that to make it generally usable. I wouldn't say this is fully achieved as long as every application is identifying the device by it's model number.

carlescufi, 2020-03-17:
This is done, see hwinfo API.

closed 2020-03-17 by carlescufi