zephyrproject-rtos/zephyr #9718: The test suite test_spi (spi_loopback) when built and run on the nrf52832_pca10040 board

nixward, 2018-08-29:
@anangl I saw you added the conf file for this board and thought you might know what is going on.  Thanks!

***** Booting Zephyr OS v1.13.0-rc1-74-g32a6fa1c5 *****
Running test suite test_spi
===================================================================
starting test - testing_spi
[general] [INF] testing_spi: SPI test on buffers TX/RX 0x20000070/0x20000090
[general] [INF] spi_complete_loop: Start
[general] [DBG] spi_async_call_cb: Polling...
[general] [ERR] spi_complete_loop: Buffer contents are different: 0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x00,
[general] [ERR] spi_complete_loop:                            vs: 0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,

    Assertion failed at /Users/nw/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:127: spi_complete_loop: (1 is true)
Buffer contents are different
FAIL - testing_spi
===================================================================
===================================================================
PROJECT EXECUTION FAILED

    Assertion failed at /Users/nw/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:333: spi_async_call_cb: (ret is true)
one or more events are not ready

anangl, 2018-09-03:
For this test a loopback must be provided, i.e. a connection between the MOSI pin and the MISO one (in the current configuration for this board these will be p0.31 and p0.30, respectively) so that the sent data can be received.

carlescufi, 2018-09-05:
Not relevant, requires additional hardware.

closed 2018-09-05 by carlescufi