zephyrproject-rtos/zephyr #9720: samples\bluetooth\mesh_demo crash with real payload

SergeSmirnov, 2018-08-29:
I use two boards nrf52840_pca10056, one as a Heartbeat subscriber (NODE_ADDR=0x0b0c), the other as a Heartbeat publisher (NODE_ADDR=0x000f).

I added two lines to the prj.conf file to use the RTT viewer:

CONFIG_RTT_CONSOLE=y
CONFIG_UART_CONSOLE=n

Minor changes to the main.c file. 
The function call board_button_1_pressed() was added to the main loop, when the example was configured as a Heartbeat subscriber.
```C
while (1) {
	k_sem_take(&tune_sem, K_FOREVER);
	board_play_tune(tune_str);
	#if NODE_ADDR != PUBLISHER_ADDR				// added line
	board_button_1_pressed();				// added line
	#endif							// added line
}
```

Also, the payload in the mesh packet in board_button_1_pressed() was added.

```C
void board_button_1_pressed(void)
{
	char str[]={"01234567"};				// added line
	NET_BUF_SIMPLE_DEFINE(msg, 3+sizeof(str) + 4);		// changed line
	struct bt_mesh_msg_ctx ctx = {
		.net_idx = net_idx,
		.app_idx = app_idx,
		.addr = target,
		.send_ttl = BT_MESH_TTL_DEFAULT,
	};
	
	/* Bind to Health model */
	bt_mesh_model_msg_init(&msg, OP_VENDOR_BUTTON);

	net_buf_simple_add_mem(&msg, str, sizeof(str));		// added line

	if (bt_mesh_model_send(&vnd_models[0], &ctx, &msg, NULL, NULL)) {
		printk("Unable to send Vendor Button message\n");
	}

	printk("Button message sent with OpCode 0x%08x\n", OP_VENDOR_BUTTON);
}
```
As a result, for each packet received by the Heartbeat subscriber, a response is sent with the string "01234567" to the Heartbeat publisher.

After several cycles, I get "***** MPU FAULT *****" on the Heartbeat subscriber board. Sometimes this happens immediately after the connection.

Subscriber's log:

```
***** Booting Zephyr OS 1.13.0-rc1 *****
Initializing...
Unicast address: 0x0b0c
Bluetooth initialized
Mesh initialized
Loading stored settings
Provisioning completed
Configuring...
Configuration complete
Subscribing to heartbeat messages
Unable to send Vendor Button message
Button message sent with OpCode 0x00c005f1
src 0x0b0c
Button message sent with OpCode 0x00c005f1
src 0x0b0c
Button message sent with OpCode 0x00c005f1
***** MPU FAULT *****
  Instruction Access Violation
***** Hardware exception *****
Current thread ID = 0x20001784
Faulting instruction address = 0x4000e004
Fatal fault in thread 0x20001784! Aborting.
```
Publisher's log:
```
***** Booting Zephyr OS 1.13.0-rc1 *****
Initializing...
Unicast address: 0x000f
Bluetooth initialized
Mesh initialized
Loading stored settings
Provisioning completed
Configuring...
Publishing heartbeat messages
Configuration complete
src 0x0b0c
src 0x0b0c
```
If the payload string is reduced by one character to "0123456", then the problem disappears. 
As I understand it, in the case of a short string there is no segmentation of packets, because the payload is 11 bytes (3 tag + 8 string) instead of 12 (3 tag + 9 string).

The issue was also in revision 1.12.99.

jhedberg, 2018-09-17:
Could you verify that your thread stacks are large enough, before we investigate this any further? Too small thread stacks is a common cause for such issues. The most likely threads that can be at fault are the Bluetooth RX thread and the system workqueue thread.

SergeSmirnov, 2018-10-17:
I am very sorry for the delay in response.

Stack sizes are default.
In the file prj.conf now added lines
```
CONFIG_INIT_STACKS=y
CONFIG_THREAD_STACK_INFO=y
```
Subscriber's log:
```***** Booting Zephyr OS 1.13.99 *****
Initializing...
Unicast address: 0x0b0c
Bluetooth initialized
Mesh initialized
Loading stored settings
Using stored settings
Subscribing to heartbeat messages
prio recv thread stack (real size 448):	unused 344	usage 104 / 448 (23 %)
recv thread stack (real size 2048):	unused 1316	usage 732 / 2048 (35 %)
prio recv thread stack (real size 448):	unused 116	usage 332 / 448 (74 %)
recv thread stack (real size 2048):	unused 1316	usage 732 / 2048 (35 %)
src 0x0b0c
Button message sent with OpCode 0x00c005f1
***** MPU FAULT *****
  Instruction Access Violation
***** Hardware exception *****
Current thread ID = 0x200017b0
Faulting instruction address = 0x4000e004
Fatal fault in thread 0x200017b0! Aborting.
adv stack (real size 512):	unused 184	usage 328 / 512 (64 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
```
Publisher's log:
```
***** Booting Zephyr OS 1.13.99 *****
Initializing...
Unicast address: 0x000f
Bluetooth initialized
Mesh initialized
Loading stored settings
Using stored settings
adv stack (real size 512):	unused 184	usage 328 / 512 (64 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
prio recv thread stack (real size 448):	unused 344	usage 104 / 448 (23 %)
recv thread stack (real size 2048):	unused 1444	usage 604 / 2048 (29 %)
src 0x0b0c
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
adv stack (real size 512):	unused 140	usage 372 / 512 (72 %)
```

jhedberg, 2018-10-17:
Is this reproducible with the latest master branch? Also, could you play with increasing the stack sizes, as well as increasing the buffer counts (since segmentation means more buffers needed) and see if that helps at all? I wasn't able to reproduce such an issue when I tried with multi-segment messages with my mesh badge app #/10210

SergeSmirnov, 2018-10-18:
The last experiment was on October 17, 2018 with the master brunch.

What parameters of stacks and buffers do you recommend to change?

jhedberg, 2018-12-13:
Are you still having this issue or were you able to solve it? Here are some config options you could play with - I recommend also looking up their documentation to understand what exactly they do.
```
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
CONFIG_CONFIG_BT_RX_STACK_SIZE=4096
CONFIG_BT_MESH_ADV_BUF_COUNT=30
CONFIG_BT_MESH_TX_SEG_MSG_COUNT=4
CONFIG_BT_MESH_TX_SEG_MSG_COUNT=4
```

SergeSmirnov, 2018-12-13:
The issue has not been solved yet.
Unfortunately, in the latest releases of the master branch, the original example does not work at all, even without my changes.
Only lines added:
```
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
CONFIG_UART_CONSOLE=n
```

Logs
```
***** Booting Zephyr OS 1.13.99 *****
Initializing...
Unicast address: 0x0b0c
Bluetooth initialized
Mesh initialized
Loading stored settings
Provisioning completed
Configuring...
Configuration complete
Subscribing to heartbeat messages
```

```
***** Booting Zephyr OS 1.13.99 *****
Initializing...
Unicast address: 0x000f
Bluetooth initialized
Mesh initialized
Loading stored settings
Provisioning completed
Configuring...
Publishing heartbeat messages
Configuration complete
```

I checked it on the boards nrf52840_pca10056, nrf52810_pca10040 and nrf51_pca10028.

jhedberg, 2018-12-14:
@SergeSmirnov what do you mean "doesn't work at all"? Your logs look normal to me.

Btw, I noticed a bug in your original code sample:
```
char str[]={"01234567"};
```
That should be:
```
char str[] = "01234567";
```

I think the first form will make a single-char array out of the pointer to the string literal (taking the lower 8 bits of the 4 byte pointer). Also, note that using sizeof() on the string will also take the terminating null character - is that intended? Also note that anything more than 15 byte payload will result in segmentation. Since sizeof("01234567") == 9 you'll get 3 + 9 + 4, i.e. 16 bytes.

SergeSmirnov, 2018-12-15:
> @SergeSmirnov what do you mean "doesn't work at all"? Your logs look normal to me.

For more information I added a debug print to the file board.h.
```
static inline void board_play_tune(const char *str)
{
	printk("%s\n", str);              // added line
}
```
This function is called in the main loop after receiving each message.

In older versions of Zephyr, a message from the publisher comes every 15 seconds. For example 23327.
```
***** Booting Zephyr OS 1.12.99 *****
Initializing...
Unicast address: 0x0b0c
Bluetooth initialized
Mesh initialized
Provisioning completed
Configuring...
Configuration complete
Subscribing to heartbeat messages
100C100D100E100F100G100A100H
100H
100H
100H
100H
100H
100H
```

```
***** Booting Zephyr OS 1.12.99 *****
Initializing...
Unicast address: 0x000f
Bluetooth initialized
Mesh initialized
Provisioning completed
Configuring...
Publishing heartbeat messages
Configuration complete
100C100D100E100F100G100A100H
```
Now this is not happening.
```
***** Booting Zephyr OS 1.13.99 *****
Initializing...
Unicast address: 0x0b0c
Bluetooth initialized
Mesh initialized
Provisioning completed
Configuring...
Configuration complete
Subscribing to heartbeat messages
100C100D100E100F100G100A100H
```

```
***** Booting Zephyr OS 1.13.99 *****
Initializing...
Unicast address: 0x000f
Bluetooth initialized
Mesh initialized
Provisioning completed
Configuring...
Publishing heartbeat messages
Configuration complete
100C100D100E100F100G100A100H
```

> Btw, I noticed a bug in your original code sample:
```
    char str[]={"01234567"};
```

> That should be:
```
    char str[] = "01234567";
```
I compiled both options and compared the binaries. They match bit by bit.

> Also note that anything more than 15 byte payload will result in segmentation. Since sizeof("01234567") == 9 you'll get 3 + 9 + 4, i.e. 16 bytes.

In my first post, I assume that the problem appears when the segmentation of messages begins.

jhedberg, 2019-03-07:
@SergeSmirnov are you still working on this and still seeing the issue, or you already resolved it?

SergeSmirnov, 2019-03-07:
Sorry, we have now switched to another task and I do not know about the current status of this issue.

carlescufi, 2019-03-19:
@jhedberg can you close this one and @SergeSmirnov can re-open if this is reproduced?

SergeSmirnov, 2019-03-19:
I agree.

closed 2019-03-19 by jhedberg