zephyrproject-rtos/zephyr #9741: tests/kernel/spinlock:kernel.multiprocessing.spinlock_bounce crashing on ESP32

spoorthik, 2018-08-31:
Latest commit on which test is run: 19b7eaaa65f6354a831f02372a9e79eab0198bc0

Execution log:
```
***** Booting Zephyr OS v1.13.0-rc2-1-g19b7eaa (d
 ESP32: starting APPCPU
ESP32: APPCPU start sequence complete
ESP32: APPCPU running
ESP32: APPCPU initialized
Running test suite spinlock
=================================================
starting test - test_spinlock_basic
PASS - test_spinlock_basic
=================================================
 starting test - test_spinlock_bounce
ESP32: starting APPCPU
```

spoorthik, 2018-08-31:
The issue is sporadic. Cannot be reproduced easily.

rljordan, 2019-03-07:
@nashif  Can we push this to future?

gnuless, 2019-03-20:
I can't even get this test to start now. Reproducibly on my setup, the console output is:

```
***** Booting Zephyr OS v1.14.0-rc1-1298-gd9e212d4f811 *****                    
ESP32: starting APPCPU                                                          
ESP32: APPCFU slaet epquonce2c):pLotdProhibited                                 
epc1=0x40084c48, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000014, depc=0x00000000
```

The culprit appears to be a NULL `_current_cpu`in `z_spin_lock_valid()`:

```
int z_spin_lock_valid(struct k_spinlock *l)
{
        if (l->thread_cpu) {
40084c43:       f88c            beqz.n  a8, 40084c56 <z_spin_lock_valid+0x1a>
40084c45:       03f490          rsr.misc0       a9
                if ((l->thread_cpu & 3) == _current_cpu->id) {
                        return 0;
40084c48:       140992          l8ui    a9, a9, 20
```
Register `a9` is allocated to `_current_cpu` here, which appears to be 0 since the load instruction is trying to load `0+20 = 0x14`, which is what the `excvaddr` is.

andyross, 2019-03-20:
At a guess, the xtensa layer is probably trying to use a spinlock before CPU initialization is finished (FWIW: _current_cpu is a macro that expands to a read from the MISC0 special register that stores the local CPU pointer).  That works, but not with the validation layer (which @dcpleung discovered yesterday has a few SMP bugs anyway).

For right now, maybe just try going into spinlock.h and disabling SPIN_VALIDATE to turn this off.

gnuless, 2019-03-20:
Disabled SPIN_VALIDATE does fix the previous problem.

Having fixed that, I can' t reproduce this problem.  I've run the test > 100 times with no failures.  I'm going to suggest that we close this bug.

andyross, 2019-03-21:
I'm sold

closed 2019-03-21 by andyross