zephyrproject-rtos/zephyr #9742: Sensor: Conversion from m/s^2 to Gs returns signed integer

jtguggedal, 2018-08-31:
The function `sensor_ms2_to_g` in the sensor API to convert from m/s^2 to Gs returns a signed integer. Is this the intended behaviour? I expected it to return a floating point number, but I might be missing something?

The function is found here:
https://github.com/zephyrproject-rtos/zephyr/blob/master/include/sensor.h#L447

MaureenHelm, 2018-09-10:
Many MCUs in Zephyr do not have a hardware floating point unit, so we tend to avoid floating point operations in general. But you're right, in this case a lot of precision is lost by returning an int. I don't think this was considered when the function was implemented.

Marking this as an API enhancement in the event we rework the sensor API.

teburd, 2025-01-13:
This function should be deprecated in the future perhaps and replaced with one that uses either fixed point or sensor_value as a return

Sensor WG 01/13/2025