zephyrproject-rtos/zephyr #9746: zephyr networking non socket client server, qemu_x86  issue

fadhel086, 2018-08-31:
Hi,
I've been testing /samples/net/echo_server and client in qemu_x86.
The server responds as below;
[echo-server] [INF] build_reply_pkt: TCP IPv6 received 536 bytes
[echo-server] [INF] build_reply_pkt: UDP IPv6 received 840 bytes
[echo-server] [INF] build_reply_pkt: UDP IPv4 received 678 bytes
[echo-server] [INF] build_reply_pkt: UDP IPv6 received 1050 bytes
[echo-server] [INF] build_reply_pkt: UDP IPv4 received 896 bytes
.....
and client as
[echo-client] [ERR] wait_reply: Data packet not received
[echo-client] [ERR] wait_reply: Data packet not received
[echo-client] [ERR] wait_reply: Data packet not received
[echo-client] [ERR] wait_reply: Data packet not received

The document mentions, that server receives and sends the packet back to client.

Would like to know, if anything related to above was discussed in qemu_x86.

The server and client are built as below
server:
cmake -DBOARD=qemu_x86 ..
make server

and client as:
cmake -DBOARD=qemu_X86 ..
make client

jukkar, 2018-08-31:
> cmake -DBOARD=qemu_X86 ..

typo here, s/qemu_X86/qemu_x86/

After using proper board the two qemu's started to work just fine. Client sends packets, server echoes them back and client receives them correctly. This with latest master.

fadhel086, 2018-08-31:
Switched to latest master and worked.
Thanx

closed 2018-08-31 by fadhel086