zephyrproject-rtos/zephyr #9748: NFFS issue after many writes by btsettings

bixivs, 2018-08-31:
Hello

I have encountered an issue using the bluetooth settings storage on an nffs partition of the flash of an nrf52832.
After many writes of the setting file the device hangs at boot trying to mount the nffs partition with the following message
```

***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0xf41ccc
***** Hardware exception *****
Current thread ID = 0x20002f04
Faulting instruction address = 0x52ee
Fatal fault in essential thread! Spinning...
```

the crash seems to happen at the instruction:

    rc = fs->mount(mp);

line 462 of the fs.c.

I replicated the same issue with a slightly modified mesh example you can find here:
http://dropbox.cblelectronics.com/6a0d2aa14337
and another board that act as a switch (configured as a generic onoff client).
The switch publish the onoff status of the client on the the same group address the onoff server are subscribed on.
After many onoff messages, If I reboot the server node, it gives the error.
I have never verified how many messages are needed to trigger the issue but 50 or more are enough.
The server node is configured to update the saved SEQ and RPL at every message.

Thanks
Daniele

carlescufi, 2018-08-31:
@bixivs thanks for the report. I am copying @jhedberg here in case there is something he can think of.

jhedberg, 2018-08-31:
Nothing comes to mind, unfortunately. It *is* worth pointing out however that pretty much all testing I've done with settings has been with the FCB backend and not NFFS.

nvlsianpu, 2018-09-04:
Thank for report. I'm off work till 10.IX. This might be manifestation of bug either in NFFS (not confirmed #9749) or during the settings destination file compression.

carlescufi, 2018-09-11:
@bixivs is this with current Zephyr master?

nvlsianpu, 2018-09-14:
@bixivs Can you verify this along with newest master as we just fixed #9749, which mighty caused this issue.

bixivs, 2018-09-14:
Sorry for the late reply!
Just tested with the current master. I get the same error.
I'm able to reproduce the issue in this way:
This is a modified mesh sample in order to use the NFFS: https://github.com/bixivs/zephyr_mesh_sample_nffs_bug.git
I use an aconno acn52832 http://aconno.de/acn52832/ as a test device, the board files are on the cbl_branch of my zephyr fork https://github.com/bixivs/zephyr.git
After a complete flash erase I program the device and provision it using the bluez meshctl tool issuing the following commands.
```
power on
security 0
discover-unprovisioned on
provision dddd0000000000000000000000000000
menu config
target 0100
appkey-add 1
bind 0 1 1000
back
```
then I issue some onoff messages
menu onoff
target 0100
onoff 0
onoff 0
onoff 0

nvlsianpu, 2018-09-19:
@bixivs Is it possible that you will provide content of nffs partition (flash adresses <0x0007a000 0x00006000) )which cause the failure (as a RAW binary file or InteHex)?

bixivs, 2018-09-19:
@nvlsianpu Sure, on the attached zip you can find two files. The "stillworking" is the content of the storage partition before the fault and the "failing" after the fault

[nffs_issue.zip](https://github.com/zephyrproject-rtos/zephyr/files/2397478/nffs_issue.zip)

nvlsianpu, 2018-09-20:
@bixivs I'm able to reproduce failure you encountered. So far it fails for your peculiar NFFS configuration (with the default one it's not failing) - I will experiment with configuration a little and give you know.

nvlsianpu, 2018-09-24:
https://github.com/apache/mynewt-nffs/issues/10

nashif, 2019-02-19:
any update on this?

nvlsianpu, 2019-02-22:
Jullabs declare to work on fixing this in near future (in a month probably) - This bug is very tough one, so even if they start today it need few week to be to fixed.

nvlsianpu, 2020-01-27:
NFFS support was removed #21793.
You can use LittleFS as FS back-end instead.

closed 2020-01-27 by nvlsianpu