zephyrproject-rtos/zephyr #9749: NRF52 : NFFS file system : use of write function returns 0 but fails

lpasteau, 2018-08-31:
Sometimes the operation fs_write() seems good when checking the return value. However the file became a file of size null and a new file with another name appears with the previous file. Then some loops later, there is an error in a return value of the write function.

When trying to modify NFFS configuration : 
- In the two cases when  the CONFIG_NFFS_FILESYSTEM_MAX_BLOCK_SIZE is higher and when the global size of the file system is higher ; the test is good during a longer time before failing, but always fails.
- If I write 24 instead of 12 bytes or if I write in two files instead of one, the issue comes sooner.

Maybe something in my configuration (or in my software) is wrong.

I reduce the time of the test in a one minute test with the  configuration below.

nrf52_pca10040.dts : 

```
/*
 * Copyright (c) 2017 Shawn Nock <shawn@monadnock.ca>
 * Copyright (c) 2017 Linaro Limited
 *
 * SPDX-License-Identifier: Apache-2.0
 */

/dts-v1/;
#include <nordic/nrf52832.dtsi>

/ {
	model = "Nordic PCA10040 Dev Kit";
	compatible = "nordic,pca10040-dk", "nordic,nrf52832-qfaa",
		     "nordic,nrf52832";

	chosen {
		zephyr,console = &uart0;
#ifdef CONFIG_MCUMGR_SMP_UART
		zephyr,uart-mcumgr = &uart0;
#endif
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
	};
};

&uart0 {
	status = "ok";
	compatible = "nordic,nrf-uart";
	current-speed = <115200>;
};
&i2c0 {
	status = "ok";
};

&i2c1 {
	status = "ok";
};

&flash0 {
	/*
	 * For more information, see:
	 * http://docs.zephyrproject.org/devices/dts/flash_partitions.html
	 */
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 0xc000>;
		};
		slot0_partition: partition@c000 {
			label = "image-0";
			reg = <0x0000C000 0x32000>;
		};
		slot1_partition: partition@3e000 {
			label = "image-1";
			reg = <0x0003E000 0x32000>;
		};
		scratch_partition: partition@70000 {
			label = "image-scratch";
			reg = <0x00070000 0xa000>;
		};

#if defined(CONFIG_FS_FLASH_STORAGE_PARTITION)
		storage_partition: partition@7a000 {
			label = "storage";
			reg = <0x0007a000 0x00002000>;
		};
#endif
	};
};
```

CMakeListt.txt : 

```
set(BOARD nrf52_pca10040)

configure_file(${CMAKE_CURRENT_SOURCE_DIR}/default.conf ${CMAKE_CURRENT_BINARY_DIR}/default.conf COPYONLY)
set(CONF_FILE  default.conf)

include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
project(NONE)

target_sources(app PRIVATE main.c)

zephyr_library_include_directories(${CMAKE_CURRENT_SOURCE_DIR}/app/include)

target_include_directories(app PRIVATE $ENV{ZEPHYR_BASE}/ext/fs/nffs/include)
target_link_libraries(app NFFS)
```

default.conf : 
```
CONFIG_BT=n
CONFIG_CONSOLE=y
CONFIG_RTT_CONSOLE=y
CONFIG_HAS_SEGGER_RTT=y
CONFIG_UART_CONSOLE=n
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_STDOUT_CONSOLE=n
CONFIG_CONSOLE_SHELL=n
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_MPU_ALLOW_FLASH_WRITE=y
CONFIG_FLASH=y
CONFIG_SOC_FLASH_NRF5=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_SOC_FLASH_NRF5_DEV_NAME="NRF5_FLASH"
CONFIG_HAS_FLASH_LOAD_OFFSET=y
CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_NFFS=y
CONFIG_NFCT_PINS_AS_GPIOS=y
CONFIG_SERIAL=y
CONFIG_UART_NRF5=y
CONFIG_GPIO_AS_PINRESET=y
CONFIG_FS_NFFS_FLASH_DEV_NAME="NRF_FLASH_DRV_NAME"
CONFIG_SOC_FLASH_NRF=y
CONFIG_FS_NFFS_NUM_BLOCKS=1024
CONFIG_FS_NFFS_NUM_INODES=1024
CONFIG_FS_NFFS_NUM_CACHE_BLOCKS=1
CONFIG_FS_NFFS_NUM_CACHE_INODES=1
CONFIG_FS_NFFS_NUM_DIRS=4
CONFIG_FS_NFFS_NUM_FILES=4
CONFIG_FS_NFFS_NUM_INODES=64
CONFIG_HEAP_MEM_POOL_SIZE=4096
CONFIG_MAIN_STACK_SIZE=4096
CONFIG_NFFS_FILESYSTEM_MAX_AREAS=12
CONFIG_NFFS_FILESYSTEM_MAX_BLOCK_SIZE=128
CONFIG_FS_FLASH_STORAGE_PARTITION=y
CONFIG_SOC_FLASH_NRF_RADIO_SYNC=y
```

main.c : 

```
#include <zephyr.h>
#include <misc/printk.h>
#include <board.h>
#include <device.h>
#include <gpio.h>
#include <flash.h>
#include <string.h>
#include <nffs/nffs.h>
#include <fs.h>
#include <stdio.h>

#define MOUNT_POINT "/nffs"
#define DIR_TEST MOUNT_POINT "/test"

#define TEST_DATA_FILE1          DIR_TEST "/test1.bin"
#define DATA_SIZE_TEST			 12
unsigned char buf_test[DATA_SIZE_TEST] = {0xA5};

void display_flash_info(void)
{
	struct fs_dir_t dir;
	struct fs_dirent dirent;
	int rc = 0;

	rc = fs_opendir(&dir, DIR_TEST);
	if (rc != 0) {
		printk("fs_opendir failed %d\n", rc);
		return;
	}

	do {
		rc = fs_readdir(&dir, &dirent);
		if (rc != 0 || dirent.name[0] == 0) {
			break;
		}
		printk("%s :  %s size %zu\n", dirent.type == 0 ? "FILE" : "DIR", dirent.name, dirent.size);
	} while (1);

	fs_closedir(&dir);
}

void test(void)
{
	int ret = 0;
	struct fs_file_t file;
	size_t size_data;

	unsigned int nb = 0;
	while(1) {

		nb++;

		if (fs_open(&file, TEST_DATA_FILE1) != 0) {
			printk("fs_open error\n");
			return;
		}
		fs_seek(&file, 0, FS_SEEK_END);

		size_data = DATA_SIZE_TEST;
		if (fs_write(&file, buf_test, size_data) != DATA_SIZE_TEST) {
			ret = -1;
			fs_close(&file);
			display_flash_info();
			printk("write error\n");
			return;
		}

		if(fs_sync(&file) != 0) {
			printk("sync error\n");
		}
//		nffs_cache_clear();
		fs_close(&file);

//		k_sleep(300);
//		display_flash_info();
//		k_sleep(300);

		printk("nb : %d\n", nb);
		if(nb == 1000) {
			break;
		}
		k_sleep(200);
	}
	display_flash_info();
}

/* NFFS work area struct */
static struct nffs_flash_desc flash_desc;

/* mounting info */
static struct fs_mount_t nffs_mnt = {
	.type = FS_NFFS,
	.mnt_point = MOUNT_POINT,
	.fs_data = &flash_desc,
};

void main(void)
{
	struct device *flash_dev;
	int res;

	flash_dev = device_get_binding(FLASH_DEV_NAME);
	if (!flash_dev) {
		printk("flash device not found\n");
		return;
	}

	/* set backend storage dev */
	nffs_mnt.storage_dev = flash_dev;

	res = fs_mount(&nffs_mnt);
	if (res < 0) {
		printk("Error mounting nffs [%d]\n", res);
		return;
	}

	res = fs_mkdir(DIR_TEST);
	if(res != 0) {
		printk("mkdir error %d\n", res);
		return;
	}
	test();
}
```

I obtain on RTT concole this display : 

```
FILE :   size 1380
FILE :   size 456
FILE :   size 72
FILE :   size 36
FILE :  test1.bin size 0
FILE :  � size 408
FILE :  � size 804
write error

```

nvlsianpu, 2018-09-04:
Thank for report. I'm off work till 10.IX. In the meantime time can you compare your configuration with configuration of tests/subsys/fs/nffs_fs_api/performance - looks like this especially this test performs many writes too.

```
void test(void)
{
	...
        ...
	while(1) {

		...
                ...
		fs_seek(&file, 0, FS_SEEK_END);   /*## append always ##*/

		size_data = DATA_SIZE_TEST;
		if (fs_write(&file, buf_test, size_data) != DATA_SIZE_TEST) {
			ret = -1;
                        ...
                        ...
			return;
		}
```
As I understand your test must fail at some point as it append data to file in each iteration.

lpasteau, 2018-09-05:
Thanks for your answer. We'll discuss after the 10 of september then.

For the future discussion : 
I tried your configuration (nrf5x.conf) with my test and I still have the issue : 
```
nb : 114
FILE :  test1.bin size 1380
nb : 115
FILE :  test1.bin size 1392
nb : 116
FILE :   size 1404
nb : 117
FILE :   size 1404
FILE :  test1.bin size 12
```
The only difference is in nrf52_pca10040.dts : 
```
#if defined(CONFIG_FS_FLASH_STORAGE_PARTITION)
		storage_partition: partition@7a000 {
			label = "storage";
			reg = <0x0007a000 **0x00002000**>;
		};
#endif
```

When I set the official config (0x6000 instead of 0x2000) the test fails : the file is removed in the loop of writing 12 bytes :  The list of files displayed is empty after the moment when test1.bin size is 9048. 

My purpose is to have in my config : reg = <0x00040000 0x00040000> for storage partition.

I agree I can't "append always". However I don't  know the overhead of the file system so I expect a return value in error when it is full. My purpose is to estimate this overhead to know the percentage of flash used in my application.

About the example with flash size at 0x6000 : Can I only write a file of 9048 bytes on a flash of 24576 bytes ?

Do you know the overhead of the file system ? Does it depend on the number of files ?

The problem of this test is that the return value is always 0, but the file is corrupted.
I did the same test with more than 1 file and there is also an issue.

Maybe my test is not correctly written and it explains the issue. Its purpose is to write 12 bytes in the file until the flash is full.

Thanks in advance for your return.

lpasteau, 2018-09-11:
Maybe the issue comes partially from the number of available "write" bewteen two block "erase" on NRF52 : 
https://devzone.nordicsemi.com/f/nordic-q-a/15798/flash-endurance-on-nrf52
- Flash write/erase cycles: 10 000.
- Amount of writes allowed in a block between erase: 181

However I tried without success a test that writes 24 bytes in a loop (instead of 12bytes in the first test). With 24 bytes each time, 170 loops are enough to write a block of 4kbytes, so less than 181.

carlescufi, 2018-09-11:
@lpasteau could you please let us know which Zephyr version you are using?

lpasteau, 2018-09-11:
I use Zephyr OS version 1.12.0

carlescufi, 2018-09-11:
@lpasteau thanks, this is a relatively recent version, but it would still be great if you could try with 1.13 (released yesterday) or master to make sure this issue is still reproduceable. If you could do that then this would be helpful so that @nvlsianpu can continue investigating.

lpasteau, 2018-09-11:
ok I try again as soon as I upgrade zephyr version !

nvlsianpu, 2018-09-12:
@lpasteau  I reproduced the issue along with zephyr master HEAD - so we are on the same line now - I continuing investigation now.

nvlsianpu, 2018-09-13:
:D today I was able to track the bug mechanism (happens when GC is performed during inode update, GC overrides file name...because uses the same buffer) tomorrow will propose a patch for that.

lpasteau, 2018-09-13:
Ok thanks a lot ! That explains why there is another file of the same size that appears but without the name...

nvlsianpu, 2018-09-14:
@lpasteau Here the fix https://github.com/apache/mynewt-nffs/pull/9 - we need to wait for merging this before fix the zephyr source, but you can use it right now.

Thanks a lot for spot the bug!

closed 2018-09-14 by carlescufi

lpasteau, 2018-09-14:
@nvlsianpu Thanks a lot the correction solves my issue :)

PR #9983 (2018-09-14): ext: nffs: bugfix of possible to corrupt filename
Origin: Apache Mynewt NFFS
URL: https://github.com/apache/mynewt-nffs/tree/master
Commit: a2f679afe6323310ab1000385f41115df7da326e
Maintained-by: External

This patch introduce version which fixes following bug:

While inode is updated it is possible that Garbage Collection
is called which caused overwrite of inode's filename as the same
buffer was used to keep the filename and copping data in GC.

This patch uses buffer on the stack for keeping filename while inode
is updated which fix the issue.

The patch cause 128 B increase of stack consum [... cut]
commit 48e6ec7b (2018-09-14): ext: nffs: bugfix of possible to corrupt filename

Origin: Apache Mynewt NFFS
URL: https://github.com/apache/mynewt-nffs/tree/master
Commit: a2f679afe6323310ab1000385f41115df7da326e
Maintained-by: External

This patch introduce version which fixes following bug:

While inode is updated it is possible that Garbage Collection
is called which caused overwrite of inode's filename as the same
buffer was used to keep the filename and copping data in GC.

This patch uses buffer on the stack for keeping filename while inode
is updated which fix the issue.

The patch cause 128 B increase of stack consu [... cut]