zephyrproject-rtos/zephyr #9762: Bluetooth mesh reliability issue?

bluetooth-mdw, 2018-09-03:
I have a test system with 66 nodes. 64 are standard nodes running the generic level server and generic onoff server models. They do not have any of the special features enabled like Relay, LPN, Friend or Proxy.

The 65th is a proxy node.

Node 66 is a proxy client which writes proxy PDUs containing generic onoff client and generic level client messages to the proxy node over a LE connection.

All nodes are in fairly close physical proximity, certainly within 2m of each other.

In testing messages addressed to groups of nodes I sometimes see that some nodes do not seem to receive my test message at all, so for example in the case of sending generic onoff set unack to a group of 16 nodes, maybe only 15 of them will switch on. I've got network retransmissions enabled in the proxy node with 4 retransmissions every n milliseconds. With n=20ms I see failures fairly frequently. With longer intervals like 50ms or 100ms I do not see failures but I do see very obvious delays in some devices responding, suggesting they are processing one of the retransmitted message copies, not the original. 

I'm wondering about RX duty cycle. Could this be the reason messages are being missed?

cvinayak, 2018-09-03:
@bluetooth-mdw could you give details of the commit hash and where I can get your application from, if it is public, else a sample that is close to reproducing your observation.

carlescufi, 2018-09-03:
Copying @jhedberg

carlescufi, 2019-08-15:
@bluetooth-mdw can you please retest with the current master, since the new controller is actually much better at avoiding packet drops?

bluetooth-mdw, 2019-08-15:
Will do when I next have a chance. FYI simply using a nRF52 based Bluetooth module instead of the nRF51 devices I used in my original tests made a big difference.

carlescufi, 2019-08-22:
@bluetooth-mdw ok thanks. Please close this if you find performance to be adequate now.

bluetooth-mdw, 2019-08-22:
Roger that.

closed 2019-08-22 by bluetooth-mdw