zephyrproject-rtos/zephyr #9767: [Coverity CID :187903] Uninitialized variables in /subsys/fs/nvs/nvs.c

mandarcthorat1, 2018-09-03:
Static code scan issues seen in File: /subsys/fs/nvs/nvs.c
 Category: Uninitialized variables
 Function: nvs_reinit
 Component: Other
 CID: 187903
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Laczen, 2018-09-03:
@nashif, how can I access the provided link ? I have never used coverity before.

I think the problem is that in theory addr could be unitialized, but this is not true when there is sector count is not zero.

Laczen, 2018-09-04:
Hi @aurel, thanks for the feedback. 

Doing the check in reinit() would give protection against someone changing fs->sector_count during program execution. However this would then need to be done in all routines. I have not studied what would happen if fs->sector_count (or any other fs->variable) is changed during execution.
All fs->variables have been declared as variables to allow easy support of external flash, ... but it is supposed that they remain constant during execution.

When it is accepted that they remain constant during execution the easiest fix is to initialize addr = 0 at the start of the reinit() routine.

aurel32, 2018-09-04:
@Laczen, indeed you are correct. Setting the address to 0 would just mean the program will crash a few lines later if `fs->sector_count` equals to 0, so I am not sure it's a good fix either, though it's probably a bit better.

I wonder if the solution there is to use an `assert(fs->sector_count != 0)`. Asserts are not enabled by default, but I don't know if they are active for the Coverity runs.

pfalcon, 2018-09-06:
> Doing the check in reinit() would give protection against someone changing fs->sector_count during program execution.

Well, there's no need to "protect" against that, it's not possible in the general case using means like checks, there're other means to that end, like memory protection.

Please keep in mind that Coverity is dumb tool, whose purpose is to make sure that *programmer* handled all the cases, not to protect against unauthorized runtime modification.

So, you always have a choice of silence Coverity warning, if it's misguided. But before that, it indeed makes sense to try to solve it, and using assert's seems like a reasonable way to do that, if you use right assert's, and Coverity will understand them.

aurel32, 2018-12-28:
Following the recent NVS changes, this issue also appear as a warning with GCC 7.3 when building with `-O2`:

```
/home/aurel32/git/zephyr/subsys/fs/nvs/nvs.c: In function 'nvs_reinit':
/home/aurel32/git/zephyr/subsys/fs/nvs/nvs.c:92:36: warning: 'addr' may be used uninitialized in this function [-Wmaybe-uninitialized]
  offset += fs->sector_size * (addr >> ADDR_SECT_SHIFT);
                              ~~~~~~^~~~~~~~~~~~~~~~~~~
/home/aurel32/git/zephyr/subsys/fs/nvs/nvs.c:606:8: note: 'addr' was declared here
  u32_t addr;
        ^~~~
```

At least with GCC it's easier to play with the code than with Coverity. I first thought that moving the checks on `fs->write_block_size` and `fs->sector_count` from `nvs_init()` to the beginning of `nvs_reinit()` would work, as `nvs_reinit()` is part of the API and can be called directly. However it doesn't work as GCC considers that `k_mutex_lock(&fs->nvs_lock, K_FOREVER)` could modify `fs->sector_count`. One way to fix that would be to move the checks after locking the mutex and unlocking it in case of error. It might be easier to just initialize `addr` to 0.

 It seems that it's not enough and the only way to really shut up GCC is to set addr to 0 in the initialization.

commit 9e8bb610 (2019-01-02): nvs: workaround a GCC and Coverity warning

Following the recent NVS changes, the following warning now appear with
GCC 7.3 when building with -O2:

ZEPHYROOT/subsys/fs/nvs/nvs.c: In function 'nvs_reinit':
ZEPHYROOT/subsys/fs/nvs/nvs.c:92:36: warning: 'addr' may be used
uninitialized in this function [-Wmaybe-uninitialized]
  offset += fs->sector_size * (addr >> ADDR_SECT_SHIFT);
                              ~~~~~~^~~~~~~~~~~~~~~~~~~
ZEPHYROOT/subsys/fs/nvs/nvs.c:606:8: note: 'addr' was declared here
  u32_t addr;
        ^~~~

This was already reported by Coverity earlier as CID:187903.

In  [... cut]
closed 2019-01-02 by carlescufi